FreeCAD/FreeCAD #17605: Sketcher: Sketch loses vertical/horizontal constraints when using Trim.

kaiwas, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Reproduction method.
1. Construct a rectangle.
2. Construct a second rectangle touching the edge of the first.
3. Use trim at the point of their contact (twice, because there are two lines at this point)
The vertical constraint remains on one line of the two formed

Video demonstration.
https://github.com/user-attachments/assets/39e55b21-883a-485d-891a-be4c41469d45

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * addFC 0.7.4
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-01:
Perhaps the problem is related to the fact that we have two lines from rectangles. The big one from the first and the small one from the second.
The small line overlaps the big one. Maybe it would be more correct to automatically delete the small one if it is "overlaid" on the big one?

depthoffocus, 2024-11-01:
I can reproduce this.

> 
> But even if we construct the second figure from three segments, we solve only half of the problem.
> 

Same is true if you draw the rectangles non-overlapping and delete, too -- only half the problem is solved. But this does suggest as you say that the broken coincident constraint happens when the right edges of the two rectangles are coincident.

https://github.com/user-attachments/assets/68b63f37-1f75-4cf0-bc02-6cfa3363e847

```
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git)
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

kaiwas, 2024-11-01:
> you say that the broken coincident constraint

Yes. That was a guess. However, it seems to have nothing to do with the loss of verticality/horizontality.
But it does have something to do with the breaking of vertices. Probably something happens to the vertex numbering during trim.

AjinkyaDahale, 2025-03-08:
Cannot reproduce with

```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40607 (Git) Snap 1329
Build date: 2025/03/08 01:10:58
Build type: Release
Branch: main
Hash: 50e80f4a03ac9eda45e2c3f80fde26d9121ad15e
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2024.9.1

```

depthoffocus, 2025-05-10:
I can no longer reproduce either the OP's original case or the variation I added.

kaiwas, 2025-05-10:
+1 It looks like the problem has been solved

luzpaz, 2025-05-10:
What PR was it fixed by?

closed 2025-05-10 by luzpaz

AjinkyaDahale, 2025-05-11:
@luzpaz probably #15811. There is a `transferToAll` on H/V constraints. It's possible, though, that some more changes had to be made later before they apply properly.