FreeCAD/FreeCAD #17606: Unable to type in scale smaller than 0.1 when number of decimals is set to 1

semhustej, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have encountered problem when trying to set scale value lower than 0.1 at several places, when FreeCAD is set to number of decimals = 1. It can be necessary to set scale at lower values than 0.1, even though there is no need to have precision higher than 1 decimal place in the project.

Places where scale precision is taken from unit precision:

- View Properties of Draft objects -> Draft -> Pattern Size
- TechDraw_View Data properties -> Base -> Scale
- TechDraw_DraftView Data properties -> Base -> Scale

I suppose there are many other 'scales' throughout the FreeCAD tools. I did not do a comprehensive research.

This behavior is confusing for new users and may also lead for user to set document do be too precise only to be able to set the correct scale.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

Proposed solution:

Not sure about the solution. One solution might be to create a new variable in the settings for unitless fields, which would be independent of unit-fields. Unfortunately, this adds complexity.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-11-01:
I see you can use expression as a workaround for setting smaller scale of TechDraw_View. There is no such option for Pattern Size option for draft objects. So the simpler solution might be to add the option to add expressions to the Patterns Size value.

maxwxyz, 2024-11-01:
So it is just a limitation in Draft?
@Roy-043 FYI

semhustej, 2024-11-01:
> So it is just a limitation in Draft? 

Yes.

Roy-043, 2024-11-01:
This is a Core problem. Expressions are not supported for View properties.

luzpaz, 2024-11-01:
> Expressions are not supported for View properties.

Rename ticket?

semhustej, 2024-11-01:
I would also classify it as a feature, not a bug.