FreeCAD/FreeCAD #17607: Expressions: There is no way to move a string to a stringlist when using expressions

kaiwas, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The documentation does not provide a way to wrap a string in a stringlist when using expressions.
The method used for formatting strings in Python does not work. However, the strings are wrapped in the preview.
[image]
[image]

A test file is attached. Remove the zip extension.
[Text_expressions.FCStd.zip](https://github.com/user-attachments/files/17598733/Text_expressions.FCStd.zip)

### Full version info

```shell
any versions
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-01:
You can use:
`list(<<a>>; <<b>>; <<c>>)`

The list function is documented:
https://wiki.freecad.org/Expressions#Object_creation_functions

Roy-043, 2024-11-01:
The Draft_Text and Draft_Label object store multi-line texts as a list of strings. But properties of this type may be used for other purposes as well. It would not be correct to say that `;` produces a linebreak, it just separates arguments here.

kaiwas, 2024-11-01:
> The Draft_Text and Draft_Label object store multi-line texts as a list of strings. But properties of this type may be used for other purposes as well. It would not be correct to say that `;` produces a linebreak, it just separates arguments here.

Yes. It's all complicated because the same recipes work differently in different places. And it's not just about this case (((
I hope that someday everything will be more stable and predictable.

Again - many thanks for the tip!

Should I close the topic?

closed 2024-11-04 by Roy-043