FreeCAD/FreeCAD #17608: Sketcher: Negative measures on circle (Version: 1.0.0RC2.38806 (Git))

CarstenH5, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Created a project from scratch.

Use XY plane
Added a circle.
Tried to use the measure tool.
Tool shows negative values.

[image]

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-01:
What measure tool did you use? The screenshot still shows an active sketch, no measure tool active. Can you share the file and steps how to reproduce this? You can also record a macro and share it here.

kaiwas, 2024-11-01:
This is very interesting because I can't seem to use a linear dimension to indicate the dimensions of a circle.

CarstenH5, 2024-11-01:
I just restarted my PC and started fresh.
I can't reproduce the error right now.
The difference is, that now when i use the Measure tool I get this dialogue at the end. 
It didn't show when the negative values were present.

[image]

CarstenH5, 2024-11-01:
This is the button I clicked for using it.
[image]

kaiwas, 2024-11-01:
> This is the button I clicked for using it.

So autosize somehow assigned a linear size to the circle? Hmm... I need to try to reproduce i

maxwxyz, 2024-11-01:
Please report back if this is reproducible again.

CarstenH5, 2024-11-01:
Ok

CarstenH5, 2024-11-01:
The first picture is misleading, sorry

maxwxyz, 2024-11-01:
Do you click on the axis line?

CarstenH5, 2024-11-01:
I click at the circle to select it.

CarstenH5, 2024-11-02:
I have made a video. 
Please take a look, if the tool behaves like expected.

https://github.com/user-attachments/assets/72a5db6d-aa0b-4518-9330-1a8f3998e692

maxwxyz, 2024-11-02:
Don't press or click on the X-axis. With the measurement tool you can click on multiple elements to dimension between them. If you click on the X-axis, it tries to dimension the midpoint with a vertical dimension to the X-axis but the midpoint is already coincident to the origin.
Try clicking on empty space to dimension only the circle.

leandropaolo1, 2025-06-13:
I have the same issue. Why would the circle be negative?

https://github.com/user-attachments/assets/14aa4f92-9c34-4ece-808d-d6010e196074

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.42099 (Git) Snap 1485
Build date: 2025/06/08 00:31:19
Build type: Release
Branch: main
Hash: 8a58d225c822d20fcf973a8de8b2cd5095d7b31e
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1

maxwxyz, 2025-06-13:
@AjinkyaDahale @FlachyJoe any idea what is going wrong here?