FreeCAD/FreeCAD #17610: Unnecessary recomputes when using CAM

sslupsky, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the CAM workbench there are frequent recomputes when there should be no recompute.  For instance, opening the Job task panel, selecting the General tab forces a recompute even though nothing has been changed.  Any time a parameter is changed in a data panel, it causes a recompute.  Thus, if you have several parameters to change you experience a recompute for every change.

There is a discussion on the forum:
https://forum.freecad.org/viewtopic.php?p=789869#p789869

CAM is not particularly performant and recomputes can take a very long time.  At the moment the only way to stop recomputes is toggle them for the entire document which is cumbersome.

Is there a way to make recomputes a separate task that run in the background?  Perhaps if the elapsed time of a recompute exceeds 1 second (or some time limit specified in the preferences), a dialog could appear asking if you want to continue or cancel the recompute.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

freckletonj, 2025-08-12:
CAM recomputes are expensive for me, thanks for bringing this up!

sslupsky, 2025-08-16:
Yeah, this is a real pain point.  I can be worse than a crash because it takes so long.  Particularly if you apply a helix ramp entry dressup.  It can literally take hours to recompute.

It would be a helpful workaround if the devs would enable the user to disable recomputes on a per operation and per dress up basis.  That might provide the devs more time to implement other changes that address the underlying issue.

freckletonj, 2025-08-16:
agreed, my recommendations:

1. if we just had a dependency graphy, and we only update downstream dependencies on changes, amazing 🤌 

2. bonus points if there were an easier way to say change multiple data attributes, and only recompute at the end of tweaking

3. bonus points for better time estimates

4. bonus points for the ability to cancel a recalculation. This would be more important than point #2. Change a data attribute, it's doing an expensive recompute? I can cancel it, and then change another property. This would require better multithreading (async in python)/multiprocessing (for spinning up OpenCascade), which would be 🔥 🔥 🔥

hansgeerdink, 2026-02-08:
Hi,
This issue makes working with CAM a real Pita.

The recalculation happens always every time one changes a value in the property window.
The recalculation happens always after changing a value in the Task panel and clicking Apply, logical / necessary.
The recalculation happens always after clicking OK in the Task panel, logical / necessary, BUT not after you had just pressed apply.

When working with 3d Surfaces and or the Waterline functions, the system either grinds to a hold or take REALLY long to recalculate. A simple padded circle (dia. 100 mm) with a fillet on the top side for instance:
Waterline.INFO: Waterline operation time is 186.16227793693542 sec.
So, every change one makes to the parameters in the property panel one has to wait for at least 3 minutes. (often a lot longer as the toolpath gets more complicated / larger)
If one changes a value in the Task panel and presses Apply to see the result before closing the Task panel one has to wait double the time, ones when pressing the Apply and one pressing the OK.

To me this seems an issue related to the general handling of the user actions.
Would it perhaps be possible to collect changes and consult / warn the user to recompute now or later?

Cheers!
Hans