FreeCAD/FreeCAD #17612: Segfault on sub-shape binder offset MAT_ListOfBisector::Current

Beep6581, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[WT32S3-86S-Solar1.zip](https://github.com/user-attachments/files/17604698/WT32S3-86S-Solar1.zip)

1. Open the project.
2. Select "body_ada > Binder_ada"
3. In the Data tab, set Offset=-2.

```
#0  0x00007fff998c56db in MAT_ListOfBisector::Current() const () from /usr/lib/libTKTopAlgo.so.7.8
#1  0x00007fff998e2b9a in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) () from /usr/lib/libTKTopAlgo.so.7.8
#2  0x00007fff998ff740 in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) () from /usr/lib/libTKTopAlgo.so.7.8
#3  0x00007fff9a851ee7 in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) () from /usr/lib/libTKBool.so.7.8
#4  0x00007fff9a8523fd in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) () from /usr/lib/libTKBool.so.7.8
#5  0x00007fffa4658db4 in ?? () from /usr/lib/libTKOffset.so.7.8
#6  0x00007fffa45873f3 in BRepOffsetAPI_MakeOffset::Perform(double, double) () from /usr/lib/libTKOffset.so.7.8
#7  0x00007fff9bd8de1a in Part::TopoShape::makeElementOffset2D(Part::TopoShape const&, double, Part::JoinType, Part::FillType, Part::OpenResult, bool, char const*) () from /home/morgan/programs/freecad/lib/Part.so
#8  0x00007fff8f8f7da9 in PartDesign::SubShapeBinder::update(PartDesign::SubShapeBinder::UpdateOption) () from /home/morgan/programs/freecad/lib/_PartDesign.so
#9  0x00007fff8f8f96c5 in PartDesign::SubShapeBinder::execute() () from /home/morgan/programs/freecad/lib/_PartDesign.so
#10 0x00007ffff4903c6a in App::DocumentObject::recompute() () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
#11 0x00007fff9bb0f58b in Part::Feature::recompute() () from /home/morgan/programs/freecad/lib/Part.so
#12 0x00007ffff488cc34 in App::Document::_recomputeFeature(App::DocumentObject*) () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
#13 0x00007ffff48917cf in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
#14 0x00007ffff589b8c7 in Gui::PropertyEditor::PropertyEditor::recomputeDocument(App::Document*) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#15 0x00007ffff589b994 in Gui::PropertyEditor::PropertyEditor::closeTransaction() () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#16 0x00007ffff589ee7d in Gui::PropertyEditor::PropertyEditor::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so

```

### Full version info

```shell
`8865450a3e14220925e0e449c0f1f79056b4fb89`
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Beep6581, 2024-11-01:
Possibly duplicates:
#9708
#8160
#6538

FEA-eng, 2026-02-28:
Still just:

`09:10:42  Binder_ada: failed to make 2D offset`

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260224 (Git shallow)
Build date: 2026/02/24 18:54:53
Build type: Release
Branch: (HEAD detached at 34ae16cd0)
Hash: 34ae16cd01b179eb9e1801591276dbc5a38669b5
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-08-28:
[This](https://github.com/FreeCAD/FreeCAD/issues/17612#issuecomment-3976642319) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

luzpaz, 2026-10-02:
@pieterhijma is this OCC related ? See https://github.com/FreeCAD/FreeCAD/issues/17612#issuecomment-2452713253