FreeCAD/FreeCAD #17613: Unable to install third party Python dependencies when developing a workbench

mroote, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When developing a workbench that needs a Python dependency specified in the workbench manifest it is not possible to install plugins that are not already added to the `ALLOWED_PYTHON_PACKAGES.txt` file. This makes local development of a plugin that needs to pull a dependency difficult as this check cannot be skipped. 

A solution to this issue could be to allow installing python packages not included in the `ALLOWED_PYTHON_PACKAGES.txt` file when development mode is enabled.  Checking the Addon Manager code it looks like adding a check in the `AddonManager._update_allowed_packages_list` method to skip this check would be enough. 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenLight/Qt default
Installed mods: 
  * cadquery2workbench
  * freecad-cadquery2-workbench 2024.11.1
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * OpenTheme 2024.9.1
  * parts_library
  * PieMenu 1.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-02:
@chennes an override option for developers ?

mwganson, 2024-11-03:
Can you try this macro and let me know if it works for you?

https://forum.freecad.org/viewtopic.php?t=91793

luzpaz, 2024-11-23:
@mroote 👆

mroote, 2024-11-24:
I'm not sure that macro really helps with this issue it looks like it's a GUI for managing the Python dependencies? I can install the package fine using the Python binary included with FreeCAD. If using this macro to install then all other users trying to install an addon with a Python dependency would need to use the macro as well.

The issue is more around testing the actual installation of addons via the addon manager. I can point the addon manager to pull from a custom repository but without editing the addon manager code I can't pull in Python dependencies to test the add on installation works correctly.