FreeCAD/FreeCAD #17618: Appimage 39100: black screen when New file is created and crash when file opened

MiqCG, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When File > New is clicked or any new file type button of the Start window are clicked, a totally black screen appears. FreeCAD is still running and you can close the file through File menu > Close.
If you try to open one of the provided examples, FreeCAD crashes
Tests made with a fresh install .cache .config .local/share FreeCAD folders deleted

https://github.com/user-attachments/assets/8ee786e2-7e70-452d-9106-dad1e306a551

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git) AppImage
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Terminal output:

`$ ./FreeCAD_weekly-builds-39100-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLShaderProgram: could not create shader program
Vertex shader for simpleShaderProg (MainVertexShader & PositionOnlyVertexShader) failed to compile
QOpenGLShader: could not create shader
Fragment shader for simpleShaderProg (MainFragmentShader & ShockingPinkSrcFragmentShader) failed to compile
Errors linking simple shader: 
QOpenGLShaderProgram: could not create shader program
Vertex shader for blitShaderProg (MainWithTexCoordsVertexShader & UntransformedPositionVertexShader) failed to compile
QOpenGLShader: could not create shader
Fragment shader for blitShaderProg (MainFragmentShader & ImageSrcFragmentShader) failed to compile
Errors linking blit shader: 
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
Could not link shader program:
 ""
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
Coin warning in cc_glglue_instance(): Error when setting up the GL context. This can happen if there is no current context, or if the context has been set up incorrectly.
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
QOpenGLTexture::setMaximumAnisotropy() requires GL_EXT_texture_filter_anisotropic
[... 94 lines cut ...]
#82  0x7f1867f8f8c3 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Widgets.so.5+0x1df
#83  0x7f1867f8f45c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Widgets.so.5+0xde6
#84  /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Widgets.so.5(+0x173def) [0x7f1867f73def]
#85  0x7f1867f940be in QWidget::event(QEvent*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Widgets.so.5+0x8e6
#86  0x7f1867f69fca in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Widgets.so.5+0xd8
#87  0x7f1869b69599 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libFreeCADGui.so+0xc9
#88  0x7f18673c3158 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5+0x8e
#89  0x7f18673c3989 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5+0x1d5
#90  /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5(+0x1f3a47) [0x7f18673f3a47]
#91  /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/./libglib-2.0.so.0(+0x5b1d5) [0x7f1865b101d5]
#92  /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/./libglib-2.0.so.0(+0x5e377) [0x7f1865b13377]
#93  /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f1865b13a8e]
#94  0x7f18673f35af in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5+0x59
#95  0x7f18673c0dda in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5+0x188
#96  0x7f18673c4ea5 in QCoreApplication::exec() from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libQt5Core.so.5+0xfb
#97  0x7f1869ac85e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAfGfMeF/usr/bin/../lib/libFreeCADGui.so+0xa41
#98  /tmp/.mount_FreeCAfGfMeF/usr/bin/freecad(+0x78b7) [0x5653e620d8b7]
#99  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f1866e6424a]
#100  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f1866e64305]
#101  /tmp/.mount_FreeCAfGfMeF/usr/bin/freecad(+0x7c2a) [0x5653e620dc2a]
`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-11-02:
This likely a packaging problem, the latest appimage is unfortunately shipping with incorrect dependencies, I'm working on fixing this

adrianinsaval, 2024-11-02:
please test the new appimage

MiqCG, 2024-11-02:
Well done, @adrianinsaval

closed 2024-11-02 by MiqCG

adrianinsaval, 2024-11-02:
is it working on multiple launches? at #17620 it is failing to launch after the first successful launch

MiqCG, 2024-11-03:
It's starting and working fine, several times, on my side.

BTW, @luzpaz , in my opinion this issue is not a duplicate of #17620

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git) AppImage
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion