FreeCAD/FreeCAD #17619: Regression in the Blender navigation style - unintuitive camera behavior when rotation center is far from the object

FEA-eng, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Everything is described in the forum thread but here's the most important part: https://forum.freecad.org/viewtopic.php?t=91595&start=10#p789527

In short, #17341 fixed some rather small bug in navigation but also brought a serious regression to the Blender style. Fortunately, #17466 mostly fixed it but there still seems to be some important regression reported in the linked forum post and confirmed by Werner: https://forum.freecad.org/viewtopic.php?t=91595&start=20#p789529

Basically, camera rotation is inconvenient when the rotation center is far from the object.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-03:
CC @Rexbas (can you take a look at this?)

Rexbas, 2024-11-03:
I am already working on this, will have a PR today. And it is not just the Blender style, but all styles combined with drag at cursor mode.

closed 2024-11-04 by yorikvanhavre

adrianinsaval, 2024-11-05:
please confirm that the behavior is as desired now

Rexbas, 2024-11-05:
We have a confirmation on the forum: https://forum.freecad.org/viewtopic.php?p=790355#p790355

PR #17639 (2024-11-04): Gui: Fix camera behavior when rotation center is far from object
Fixes #17619 which is a second regression caused by #17341 and also fixes a regression in turntable navigation speed caused by #17466.

This is the behavior after this PR gets merged:

https://github.com/user-attachments/assets/9634e5c4-8aac-4f2d-9dcb-47d314538a6c

The behavior in build 38991 and 39092 can be seen here: https://forum.freecad.org/viewtopic.php?p=789527#p789527.

With this PR the behavior is more like it was in build 38991 but with slightly different behavior when the mouse starts a rotation while above an object.  [... cut]