FreeCAD/FreeCAD #17620: Segmentation Fault in latest Dev.

GAZ082, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cant run latest FC. Please see attached console output.
[output.txt](https://github.com/user-attachments/files/17607314/output.txt)

**RC2 works fine.**

### Full version info

```shell
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-02:
@adrianinsaval could you take a look please?

adrianinsaval, 2024-11-02:
https://github.com/FreeCAD/FreeCAD/issues/17618#issuecomment-2453033910

adrianinsaval, 2024-11-02:
new appimage up, please try

GAZ082, 2024-11-02:
Now lauches! Thanks!

GAZ082, 2024-11-02:
No, wait a second, now it does not. Let me investigate.

GAZ082, 2024-11-02:
First run:
```
gabriel@xiridio:tmp$ ./FreeCAD_weekly-builds-39100-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
Cannot read termcap database;
using dumb terminal settings.
free(): invalid pointer
/tmp/.mount_FreeCAgKiOMi/AppRun: line 41: 75318 Aborted                 (core dumped) "${MAIN}" "$@"
```
Second run:
```
gabriel@xiridio:tmp$ ./FreeCAD_weekly-builds-39100-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory
Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7c3048a6a1d0]
#1  /usr/lib/libc.so.6(__libc_free+0x25) [0x7c3048ad2575]
#2  /tmp/.mount_FreeCAAoiOep/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7c303323e68e]
#3  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7c304aecc1cc]
#4  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7c304ae025fb]
#5  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7c304aded6ab]
#6  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#7  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7c304adfb376]
#8  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7c304ae2a1af]
#9  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7c304ae299d7]
#10  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7c304ae3eb5e]
#11  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7c304adf3d46]
#12  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7c304aded6ab]
#13  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#14  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7c304add5e24]
#15  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7c304ad02bc7]
#16  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x7c304ae9fbad]
#17  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7c304ae9f5a4]
#18  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x7c304aeb610e]
#19  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7c304adf3d46]
#20  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7c304aded6ab]
#21  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#22  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7c304adfb376]
#23  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7c304ae2a1af]
#24  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7c304ae299d7]
#25  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7c304ae3eb5e]
#26  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7c304adf3d46]
#27  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7c304aded6ab]
#28  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#29  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7c304add5e24]
#30  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7c304ad02bc7]
#31  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#32  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7c304add5e24]
#33  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7c304add5ab0]
#34  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7c304a342d6a]
#35  /tmp/.mount_FreeCAAoiOep/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7c301ed572f4]
#36  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7c304aecc1cc]
#37  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7c304ae025fb]
#38  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7c304aded6ab]
#39  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7c304ae0bc1f]
#40  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7c304adfb376]
#41  /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7c304ae2a1af]
[... cut ...]
#106  0x7c304b79c63d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7c304b6c809f in Gui::Application::runApplication() from /tmp/.mount_FreeCAAoiOep/usr/bin/../lib/libFreeCADGui.so+0x4ff
#108  /tmp/.mount_FreeCAAoiOep/usr/bin/freecad(+0x78b7) [0x5b452184e8b7]
#109  /usr/lib/libc.so.6(+0x25e08) [0x7c3048a52e08]
#110  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7c3048a52ecc]
#111  /tmp/.mount_FreeCAAoiOep/usr/bin/freecad(+0x7c2a) [0x5b452184ec2a]
```

adrianinsaval, 2024-11-02:
huh that's weird, same for me runs fine first time but crashes a second time

luzpaz, 2024-11-03:
Crash
<details><summary>Crash dump</summary>

```
Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f196a66a1d0]
#1  /usr/lib/libc.so.6(__libc_free+0x25) [0x7f196a6d2575]
#2  /tmp/.mount_FreeCAecNfOj/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7f19542c868e]
#3  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f196cacc1cc]
#4  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f196ca025fb]
#5  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#6  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#7  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f196c9fb376]
#8  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f196ca2a1af]
#9  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f196ca299d7]
#10  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f196ca3eb5e]
#11  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f196c9f3d46]
#12  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#13  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#14  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f196c9d5e24]
#15  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f196c902bc7]
#16  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x7f196ca9fbad]
#17  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f196ca9f5a4]
#18  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x7f196cab610e]
#19  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f196c9f3d46]
#20  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#21  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#22  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f196c9fb376]
#23  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f196ca2a1af]
#24  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f196ca299d7]
#25  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f196ca3eb5e]
#26  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f196c9f3d46]
#27  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#28  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#29  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f196c9d5e24]
#30  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f196c902bc7]
#31  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#32  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f196c9d5e24]
#33  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7f196c9d5ab0]
#34  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7f196c684d6a]
#35  /tmp/.mount_FreeCAecNfOj/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7f1944d572f4]
#36  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f196cacc1cc]
#37  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f196ca025fb]
#38  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#39  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#40  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f196c9fb376]
#41  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f196ca2a1af]
#42  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f196ca299d7]
#43  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f196ca3eb5e]
#44  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f196c9f3d46]
#45  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f196c9ed6ab]
#46  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f196ca0bc1f]
#47  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x13b) [0x7f196c9d589b]
#48  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyImport_Import+0x19b) [0x7f196ca3e5fb]
#49  /tmp/.mount_FreeCAecNfOj/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModule+0x18) [0x7f196cabb018]
[... cut ...]
#108  /tmp/.mount_FreeCAecNfOj/usr/bin/freecad(+0x78b7) [0x55971c7008b7]
#109  /usr/lib/libc.so.6(+0x25e08) [0x7f196a652e08]
#110  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f196a652ecc]
#111  /tmp/.mount_FreeCAecNfOj/usr/bin/freecad(+0x7c2a) [0x55971c700c2a]
```

</details>

```
Manjaro Linux
FreeCAD 1.1.0 Revision: 39100 (Git)

Library versions:
boost    1_84
Coin3D   4.0.3
Eigen3   3.4.0
OCC      7.8.1
Qt       5.15.15
Python   3.11.10
PySide   5.15.15
shiboken 5.15.15
SMESH    9.8.0.2
VTK      9.3.0
xerces-c 3.2.5
```

adrianinsaval, 2024-11-03:
now I'm confused, if I create an appimage locally it works

GAZ082, 2024-11-03:
If you try to run it 6 or 7 times it end up starting.

adrianinsaval, 2024-11-03:
On a different computer it works without issues for me, could you post details about your system? If you can get it to run again pleas post full freecad version info too.

For me it works under kde x11 on archlinux on lenovo with intel cpu and nvidia gpu, fails on kde wayland on fedora on an macbookpro

GAZ082, 2024-11-03:
OS: Arch Linux (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git) AppImage
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29
  * ThreadProfile 1.92.0
  * OpenTheme 2024.9.1

GAZ082, 2024-11-03:
@adrianinsaval on Wayland also works, after trying 6 or 7 times.

peterwwillis, 2024-11-03:
Hello, same issue, below is my deets

```
FreeCAD 1.1.0 Revision: 39100 (Git)

Library versions:
boost    1_84
Coin3D   4.0.3
Eigen3   3.4.0
OCC      7.8.1
Qt       5.15.15
Python   3.11.10
PySide   5.15.15
shiboken 5.15.15
SMESH    9.8.0.2
VTK      9.3.0
xerces-c 3.2.5
```

Crash dump: <details>
```
peterw@my-mbp:~$ ./Downloads/FreeCAD_weekly-builds-39100-conda-Linux-aarch64-py311.AppImage --version --verbose
xkbcommon: ERROR: failed to add default include path /tmp/cirrus-ci-build/conda/linux/AppDir/usr/share/X11/xkb
qt.qpa.wayland: failed to create xkb context
Program received signal SIGSEGV, Segmentation fault.
#0  linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffff91e94840]
#1  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0xffff87e0e6c0]
#2  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x21434) [0xffff87e11434]
#3  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x58) [0xffff87e10988]
#4  0xffff87fc128c in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xe0
#5  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6ec0) [0xffff87ce6ec0]
#6  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6438) [0xffff87ce6438]
#7  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xb418) [0xffff87e6b418]
#8  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6cd4) [0xffff87e66cd4]
#9  /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x7c) [0xffff87e680e0]
#10  0xffff87fc58e0 in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x38
#11  0xffff87fcbc78 in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_FreeCAjDBJmp/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xec
#12  0xffff8dcc59b0 in QWindowPrivate::setVisible(bool) from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Gui.so.5+0x198
#13  0xffff90addaf8 in QWidgetPrivate::show_sys() from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Widgets.so.5+0x2a0
#14  0xffff90ae3bdc in QWidgetPrivate::show_helper() from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Widgets.so.5+0x12c
#15  0xffff90ae5b94 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Widgets.so.5+0x2f8
#16  0xffff90c09db4 in QDialog::setVisible(bool) from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Widgets.so.5+0xa4
#17  0xffff90c09668 in QDialog::exec() from /tmp/.mount_FreeCAjDBJmp/usr/bin/../lib/libQt5Widgets.so.5+0xd4
#18  /tmp/.mount_FreeCAjDBJmp/usr/bin/freecad(+0x7a68) [0xaaaaeed87a68]
#19  /lib64/libc.so.6(+0x30a1c) [0xffff8d4d0a1c]
#20  /lib64/libc.so.6(__libc_start_main+0x9c) [0xffff8d4d0afc]
#21  /tmp/.mount_FreeCAjDBJmp/usr/bin/freecad(+0x7bf4) [0xaaaaeed87bf4]
```
</details>

Alternate crash dump using different env vars: <details>
```
peterw@my-mbp:~$ QT_QPA_PLATFORM="xcb" WaylandEnable=false ./Downloads/FreeCAD_weekly-builds-39100-conda-Linux-aarch64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffff2b1dc840]
#1  /lib64/libc.so.6(__libc_free+0x44) [0xffff266f8d30]
#2  /tmp/.mount_FreeCAoOpjDo/usr/lib/python3.11/lib-dynload/readline.cpython-311-aarch64-linux-gnu.so(PyInit_readline+0x208) [0xffff103d5b78]
#3  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x274368) [0xffff2ae34368]
#4  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x2708e8) [0xffff2ae308e8]
#5  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x1a5e10) [0xffff2ad65e10]
#6  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x140c) [0xffff2acbc84c]
#7  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x246a94) [0xffff2ae06a94]
#8  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x158e6c) [0xffff2ad18e6c]
#9  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0xc8) [0xffff2ad190c8]
[... 86 lines cut ...]
#96  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x246a94) [0xffff2ae06a94]
#97  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0xffff2ae06b4c]
#98  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x23d6e8) [0xffff2adfd6e8]
#99  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x1a55b8) [0xffff2ad655b8]
#100  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(PyObject_Vectorcall+0x48) [0xffff2ad18ae8]
#101  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6ec8) [0xffff2acc2308]
#102  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x246a94) [0xffff2ae06a94]
#103  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0xffff2ae06b4c]
#104  /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libpython3.11.so.1.0(+0x28f0cc) [0xffff2ae4f0cc]
[... cut ...]
#110  0xffff28545cac in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libFreeCADGui.so+0xcc
#111  0xffff2840379c in Gui::Application::runApplication() from /tmp/.mount_FreeCAoOpjDo/usr/bin/../lib/libFreeCADGui.so+0x50c
#112  /tmp/.mount_FreeCAoOpjDo/usr/bin/freecad(+0x6b60) [0xaaab9ffe6b60]
#113  /lib64/libc.so.6(+0x30a1c) [0xffff26680a1c]
#114  /lib64/libc.so.6(__libc_start_main+0x9c) [0xffff26680afc]
#115  /tmp/.mount_FreeCAoOpjDo/usr/bin/freecad(+0x7bf4) [0xaaab9ffe7bf4]
```
</details>

OS: Asahi Linux, see screenshot for info
[image]

GAZ082, 2024-11-05:
Still happening with 39127

Again, if I try several times it ends up launching.

EchoJunction, 2024-11-05:
Same thing happens to me. I'm on Arch Linux with Wayland. I tried launching it 10 times in a row but couldn't get it to launch. This is my console output:

```
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)                                                                                                                     
(C) 2001-2024 FreeCAD contributors                                                                                                                            
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.                                                                          
                                                                                                                                                              
xkbcommon: ERROR: failed to add default include path /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/X11/xkb                     
/home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: No such file or directory                                     
QOpenGLFunctions created with non-current context                                                                                                             
Program received signal SIGSEGV, Segmentation fault.                                                                                                          
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x771b3d46a1d0]                                                                                                             
#1  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0x771b3915f280]                                            
#2  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x1e775) [0x771b39161775]                                                     
#3  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x47) [0x771b39160f37]                              
#4  0x771b3878d526 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xda                                                                                                          
#5  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6a4a) [0x771b39d59a4a]                                                          
#6  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/./libffi.so.8(+0x5fea) [0x771b39d58fea]                                                          
#7  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xa573) [0x771b3c02c573]                                                 
#8  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6950) [0x771b3c028950]                                                 
#9  /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x6b) [0x771b3c029a3b]                  
#10  0x771b387919b8 in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x34                                                                                                                                                        
#11  0x771b3879771a in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_freecaEaMkcN/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x124                                                                                                                                                       
#12  0x771b3df100d2 in QWindowPrivate::setVisible(bool) from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libQt5Gui.so.5+0x1b8                                     
#13  0x771b3e58ba7c in QWidgetPrivate::show_sys() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libQt5Widgets.so.5+0x2aa                                       
#14  0x771b3e591847 in QWidgetPrivate::show_helper() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libQt5Widgets.so.5+0x123                                    
#15  0x771b3e593672 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libQt5Widgets.so.5+0x210                                 
#16  0x771b406550a1 in Gui::MainWindow::startSplasher() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libFreeCADGui.so+0x2a1                                   
#17  0x771b4019a592 in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libFreeCADGui.so+0x142                          
#18  0x771b4019c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libFreeCADGui.so+0x6d                                  
#19  0x771b400c8092 in Gui::Application::runApplication() from /tmp/.mount_freecaEaMkcN/usr/bin/../lib/libFreeCADGui.so+0x4e2                                 
#20  /tmp/.mount_freecaEaMkcN/usr/bin/freecad(+0x78b7) [0x60ae7ed688b7]                                                                                       
#21  /usr/lib/libc.so.6(+0x25e08) [0x771b3d452e08]                                                                                                            
#22  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x771b3d452ecc]                                                                                              
#23  /tmp/.mount_freecaEaMkcN/usr/bin/freecad(+0x7c2a) [0x60ae7ed68c2a]
```

TheRealOne78, 2024-11-05:
It's not about segfaulting AppImage but rather GLX not loading.

I changed my nVidia card with an AMD GPU and the appimages from FreeCAD nolonger run:

```
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
./AppRun: line 31: 24916 Aborted                 "${MAIN}" "$@"
```
This happens on Gentoo and Fedora 41.

For r39127 I also had to extract the appimage due to the following:
```
fuse: failed to exec fusermount: Permission denied

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
open dir error: No such file or directory
```

This is even on 0.21.2:
```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
/tmp/.mount_FreeCA1i0Vs2/AppRun: line 43: 28594 Aborted                 ${MAIN} "$@
```

I thought that since it's about AppImage packaging, this issue would rather fit here.

GAZ082, 2024-11-05:
@EchoJunction I'm also with Arch. Both Wayland and XORG fail but if i try 7 or 8 times, it finally launches. I run it on the console.

What a freak bug.

byteofsoren, 2024-11-06:
I tried `$ watch -n 2 freecad.appimage` but it spitted out the same error all the time.

```
Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x79ff6926a1d0]
#1  /usr/lib/libc.so.6(__libc_free+0x25) [0x79ff692d2575]
#2  /tmp/.mount_freecaLnhBlm/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x79ff5bf2c68e]
#3  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x79ff6b6cc1cc]
#4  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x79ff6b6025fb]
#5  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#6  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#7  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x79ff6b5fb376]
#8  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x79ff6b62a1af]
#9  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x79ff6b6299d7]
#10  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x79ff6b63eb5e]
#11  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x79ff6b5f3d46]
#12  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#13  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#14  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x79ff6b5d5e24]
#15  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x79ff6b502bc7]
#16  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x79ff6b69fbad]
#17  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x79ff6b69f5a4]
#18  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x79ff6b6b610e]
#19  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x79ff6b5f3d46]
#20  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#21  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#22  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x79ff6b5fb376]
#23  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x79ff6b62a1af]
#24  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x79ff6b6299d7]
#25  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x79ff6b63eb5e]
#26  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x79ff6b5f3d46]
#27  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#28  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#29  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x79ff6b5d5e24]
#30  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x79ff6b502bc7]
#31  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#32  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x79ff6b5d5e24]
#33  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x79ff6b5d5ab0]
#34  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x79ff6a7cdd6a]
#35  /tmp/.mount_freecaLnhBlm/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x79ff323572f4]
#36  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x79ff6b6cc1cc]
#37  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x79ff6b6025fb]
#38  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#39  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#40  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x79ff6b5fb376]
#41  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x79ff6b62a1af]
#42  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x79ff6b6299d7]
#43  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x79ff6b63eb5e]
#44  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x79ff6b5f3d46]
#45  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x79ff6b5ed6ab]
#46  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x79ff6b60bc1f]
#47  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x13b) [0x79ff6b5d589b]
#48  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_Import+0x19b) [0x79ff6b63e5fb]
#49  /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModule+0x18) [0x79ff6b6bb018]
[... cut ...]
#106  0x79ff6bf9c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x79ff6bec8092 in Gui::Application::runApplication() from /tmp/.mount_freecaLnhBlm/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_freecaLnhBlm/usr/bin/freecad(+0x78b7) [0x588d6873d8b7]
#109  /usr/lib/libc.so.6(+0x25e08) [0x79ff69252e08]
#110  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x79ff69252ecc]
#111  /tmp/.mount_freecaLnhBlm/usr/bin/freecad(+0x7c2a) [0x588d6873dc2a]

```

GAZ082, 2024-11-08:
Same with 39141

byteofsoren, 2024-11-09:
Yep. Just checked and I have the same problem with 39141.

maxwxyz, 2024-11-09:
@adrianinsaval is this still an issue with the weekly builds or will a release version fail too?

byteofsoren, 2024-11-09:
Nope. 
Only the two previous weekly builds crashes like this. 
I'm running RC2 AppImage right now with out any difficulty.

```
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.4.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Gridfinity 0.6.0
```

benj5378, 2024-11-09:
Seems like duplicate of #17153. Can you try @xtemp09 proposal: temporarily rename ~/.local/ folder?

@GAZ082 @byteofsoren

adrianinsaval, 2024-11-10:
I don't think it's the same, in your report we can clearly see a local installation is messing with the appimage's python:
```
#6  /home/bensay/.local/lib/python3.11/site-packages/PySide2/QtCore.abi3.so(+0x251078) [0x7f597cc51078]
#7  /home/bensay/.local/lib/python3.11/site-packages/PySide2/QtCore.abi3.so(PyInit_QtCore+0x4f) [0x7f597cccbe3f]
```
there's nothing like that in these reports.

adrianinsaval, 2024-11-10:
new appimage up, please test. I reverted back to the previous set of dependencies and it seems to work at least on my previously failing machine. In any case this seems to be a purely appimage/conda packaging issue so I will close this in favor of the duplicate in the relevant repo: https://github.com/FreeCAD/FreeCAD-Bundle/issues/332 please continue the conversation there

closed 2024-11-10 by adrianinsaval