FreeCAD/FreeCAD #17621: Regression: High CPU utilization on the start screen

semhustej, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the latest build I get high CPU utilization (seems like 1 core is running at 100%) at the start screen doing nothing. When a file is opened, the CPU usage goes to normal level less than 1%.

Steps to reproduce:

- Start FreeCAD
- CPU runs at high level

I am not able to reproduce this behavior with 1.0RC2. I think (but I am not completely sure), that this started with 1.1.0dev.39100 (Git) build.

I tried deleting my system.cfg and user.cfg files with no change.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Core

### Anything else?

Not sure if it is related, but some buttons on the start tab seem to be set in different style (rounded corners), than what I have set-up (Classic).

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-11-02:
This is already reported and [PR](https://github.com/FreeCAD/FreeCAD/pull/17565) created

semhustej, 2024-11-02:
> This is already reported and [PR](https://github.com/FreeCAD/FreeCAD/pull/17565) created

OK, thank you. I am closing this issue.

closed 2024-11-02 by semhustej