FreeCAD/FreeCAD #17623: Assembly: Joint Reverse button no longer works

Roy-043, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a joint, or editing a joint, the Reverse button no longer works.
1. Open the attached [file](https://github.com/user-attachments/files/17607502/assy.FCStd.zip) (remove the .zip extension).
2. Activate the assembly.
3. Double-click the last joint.
4. Try using the Reverse button.
5. Result: the button has no effect.

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-02:
Confirmed, **only in latest weekly**. 
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Cannot reproduce in the one from Monday:

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

@PaddleStroke FYI

MiqCG, 2024-11-03:
Take into account that probably the last weekly (rev 39100) is not totally fine: first [#17618](https://github.com/FreeCAD/FreeCAD/issues/17618) and then [#17620](https://github.com/FreeCAD/FreeCAD/issues/17620)

closed 2024-11-04 by yorikvanhavre