FreeCAD/FreeCAD #17624: Quick Measure: Error when dropping a Part_line on an assembly

Roy-043, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Part Line with the [Part_Primitives](https://wiki.freecad.org/Part_Primitives) tool.
2. Create an assembly.
3. Drop the line on the assembly.
4. Result: Error message.

`16:57:25  <QuickMeasure> QuickMeasure.cpp(97): 000000BE092551C0 : Standard_TypeMismatch: TopoDS::Edge`

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-02:
@PaddleStroke FYI

PaddleStroke, 2024-11-06:
First observation, the same happens with App::Part. So not related to assembly.

PaddleStroke, 2024-11-06:
So yes a QuickMeasure issue.
It happens when you select a Part::Line in the tree. Interestingly it does not happen if you select the line in the view. Here's the different addSelection : 

View : 
>>> # Gui.Selection.addSelection('Unnamed','Part','Line.Edge1',6.05477,6.05477,6.05477)
Tree:
>>> # Gui.Selection.addSelection('Unnamed','Part','Line.')

So it appears that those Part::Line objects have both a name and a subname.

So it comes from the fact that 
`Part::Feature::getShape(*obj, (*subEl).c_str(), true);` in Measurement.cpp is finding a TopoEdge subshape for obj = Part and sub = 'Line.'
Interestingly, getShape does not find a TopoEdge for obj = Line and empty sub.
Which is why this happens only when the Part::Line is in a App::Part.

I'd say that there's at least an inconsistency in `getShape`. I don't know if it is intended to find the edge shape from the Part::Line object. But if so then it should always find it.

closed 2024-11-11 by yorikvanhavre