FreeCAD/FreeCAD #17625: Assembly: "Solve failed: iterNo > iterMax" warning when creating Cylindrical Joint

Roy-043, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open the attached [file](https://github.com/user-attachments/files/17607634/assy.FCStd.zip) (remove the .zip extension).
2. Activate the assembly.
3. Select the outer circular edge of the hollowed out cylinder facing you.
4. Select Axel2 (select the edge in the 3D View).
5. Invoke [Assembly_CreateJointCylindrical](https://wiki.freecad.org/Assembly_CreateJointCylindrical).
6. Result: warning: `17:23:09  267.743 <Assembly> AssemblyObject.cpp(175): Solve failed: iterNo > iterMax`.
7. Repeat the test but reverse the selection order.
8. Result: no warning.

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

aiksiongkoh, 2024-11-08:
I start FreeCAD 1.0.0RC2.
Do steps 1 to 4.
At step 5, all the Assembly icons are gray.
Please advise.
Thanks.

luzpaz, 2024-11-08:
> I start FreeCAD 1.0.0RC2.

@aiksiongkoh the ticket is for the weekly build. Can you try with the weekly ?

aiksiongkoh, 2024-11-12:
[image]

filippopairotti, 2024-11-24:
i'm still having this problem in the recent 1.0, it's coming in the same way. on my side all the assembly "explode after this"

aiksiongkoh, 2024-11-25:
In FreeCAD Preferences Assembly, turn on debug. Make assembly to reproduce the bug. Look for latest *.asmt file and upload it. Thanks

j-hap, 2024-11-30:
judging from the screenshot, the assembly seems to have lost the focus ("active object") and therefore no assembly action is possible

PaddleStroke, 2025-11-24:
@Roy-043 can you please update the issue title ? As current title does not really reflect the root issue.

aiksiongkoh, 2026-09-30:
Should be closed.

FEA-eng, 2026-09-30:
I can still reproduce it in the latest weekly.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

aiksiongkoh, 2026-09-30:
This is a mathematical problem.CylindricalJoint requiresMarkerJ origin to be along MarkerI z-axis.MarkerJ x-axis and y-axis to be perpendicular to MarkerI z-axis.MarkerI is the reference to constraint MarkerJ.When MarkerI is on a stable part which is fixed or well constrained, the solver has an easier time adjusting MarkerJ to solve.But when MarkerI is on a free moving part, the solver has a harder time because it is moving the reference as well.So, the best practice is to always put MarkerI of joints on the more stable part.Also, as always, the closer the initial guess is to the solution the better.My comment from one year ago. Aik-Siong Koh

Image: Image (view on web)