FreeCAD/FreeCAD #17626: Assembly: Simple offset tab has no input widget

Roy-043, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Simple offset tab in the task panel has no input widget. See image. In previous versions you can enter a 'simple' offset.

[image]

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-02:
@PaddleStroke is this the expected offset dialog?

PaddleStroke, 2024-11-02:
Yes because cylindrical did not have rotation or offset before. So the simple mode is empty.
But this is not ideal I know. Any suggestion ? Should we always show rotation and offset even for joints that don't need them ?

Roy-043, 2024-11-04:
It is indeed not a regression. I mixed up the revolute and the cylindrical joint.

I see 2 options:
* Hide the Simple tab.
* A text on the Simple tab explaining that a simple offset is not available as it makes no sense for this joint.

Roy-043, 2024-11-04:
FYI: For the Distance joint the Simple tab is also empty. That is a regression AFAICT.

kadet1090, 2024-11-04:
This looks like it should have an input so this is a serious UX issue as users will be confused. In that case the whole section should not be visible.

kadet1090, 2024-11-04:
Showing useless controls is well, useless and wastes user time and energy - this is also not a good solution. This suggest that the whole concept of using tab box here is flawed and instead we should use another UI control like collapsible boxes.

My initial idea is to change it from using tab box to two collapsible group boxes. This control is not yet available, but I constantly see a need for it so I'll implement it soon:
[image]

We can program them in a way that only one section is visible, but I don't think that it is really necessary.

Clickable prototype:
https://www.figma.com/proto/nDkUKcH1YgriUSRG2EInYj/FreeCAD-Dialogs?node-id=158-2817&node-type=frame&t=q8n9tVXNA4siMxCT-1&scaling=min-zoom&content-scaling=fixed&page-id=0%3A1&starting-point-node-id=158%3A2817&show-proto-sidebar=1

PaddleStroke, 2024-11-05:
What about a checkbox?
[image]
This way we don't need to add a custom widget type. Seems to me the best solution.

kadet1090, 2024-11-05:
Yep, this is fine for now, we can revisit it later if such widget is introduced. However you should stick with "Show advanced offsets" as label - it is more clear.

closed 2024-11-09 by chennes