FreeCAD/FreeCAD #17628: BIM: Export to IFC fails due to new ifcopenshell version

Roy-043, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Export an Arch Wall to IFC.
Result: Warning, see below.

`ifcopenshell.version => '0.0.0'  # seems strange?`

Solution: I think we need to import the guid module now: `Import ifcopenshell.guid`

Warning:
```
19:56:51  pyException: Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "D:\BKG_Tmp\FreeCAD_0.22\Mod\BIM\importers\exportIFC.py", line 305, in export
    contextCreator = exportIFCHelper.ContextCreator(ifcfile, objectslist)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\BKG_Tmp\FreeCAD_0.22\Mod\BIM\importers\exportIFCHelper.py", line 133, in __init__
    self.project = self.createProject()
                   ^^^^^^^^^^^^^^^^^^^^
  File "D:\BKG_Tmp\FreeCAD_0.22\Mod\BIM\importers\exportIFCHelper.py", line 193, in createProject
    return self.createAutomaticProject()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\BKG_Tmp\FreeCAD_0.22\Mod\BIM\importers\exportIFCHelper.py", line 198, in createAutomaticProject
    self.getProjectGUID(),
    ^^^^^^^^^^^^^^^^^^^^^
  File "D:\BKG_Tmp\FreeCAD_0.22\Mod\BIM\importers\exportIFCHelper.py", line 217, in getProjectGUID
    return ifcopenshell.guid.new()
           ^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: module 'ifcopenshell' has no attribute 'guid'
```

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-03:
@yorikvanhavre could you please take a look?

yorikvanhavre, 2024-11-04:
IIRC in IfcOpenShell 0.8 now on there is no more implicot imports... So if one wants to use the guid module, one needs to `from ifcopenshell import guid` first. I'll issue a PR here

PR #17687 (2024-11-05): BIM: Fixed imports of IfcOpenShell 0.8
fixes #17628
commit 928decce (2024-11-05): BIM: Fixed imports of IfcOpenShell 0.8 - fixes #17628
commit 3b7bae03 (2024-11-05): BIM: Fixed imports of IfcOpenShell 0.8 - fixes #17628
closed 2024-11-05 by yorikvanhavre