FreeCAD/FreeCAD #17629: Sketch/Attach Sketch... will cause a Sketch to fail with "The graph must be a DAG." message

Buffalo1John, 2024-11-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create New empty document
Create Body
Create Sketch
Select XY-plane         NOTE this could be any plane, it does not matter
Draw Circle or Rectangle or anything             NOTE that the contents of the sketch does not matter
Close sketch
Select Sketch command tab then Attach Sketch...
Select the Sketch from the list then OK               NOTE there is only one sketch
Select Object's XY (suggested) then OK            NOTE I tried the other choices of plane, all have same results

After OK, I see two new errors in the report view:
13:58:20  1494.04 <App> Document.cpp(2540): The graph must be a DAG.
13:58:20  1494.05 <App> Document.cpp(2900): Unnamed#Sketch still touched after recompute
[image]

Why am I doing Attach Sketch - well, I found this problem when importing a DXF file and converting it to a Sketch in the DRAFT workbench. I thought it would be a good idea to attach a sketch to a plane if it was the first sketch of the Body. So, I tried to find a simpler way to reproduce the error message.

I also found that I can make the error go away if I selected Attachment... in the Data tab and then press Reset in the dialog and you get:
[image]

and then the DAG is gone. However, if I try to do Attach Sketch... again, the DAG error returns.

If now, with the DAG error, I select Map Mode in the Data tab, I see:
[image]

Just a guess, but that doesn't look right - attached to a "Wire" "Sketch"? 
If I force the reference 1 to be Selecting... and then select the Body/Origin/XY-plane the DAG goes away.

NOTE: this happens on 1.1.0dev and 0.21.2 (in 0.21.2 the command name was "Map sketch to face...")

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-03:
The way that command is designed to work is a bit odd, but here it is.  First you select the face, plane, etc., that want to attach a sketch to, then you execute the command and select the sketch to attach to that object.  It's counterintuitive because you would naturally expect to do it in the reverse order, which is to say first select the sketch, and then in the dialog select the object to attach the sketch to.

So, when you select the sketch first, and then execute the command it opens the sketch selection combo box, which contains all the sketches in the current document.  When you select the sketch that you already selected it sets you up to attach the sketch to itself, which is a big no no because then you have a circular reference, which is what the cryptic graph must be a DAG message is telling you.  DAG stands for Directed Acyclic Graph, Acyclic meaning, not cyclic, in other words, no circular dependencies.

Obviously, the sketch already selected should not appear in the combo box.  It is valid to attach a sketch to another sketch, but not to itself.

Buffalo1John, 2024-11-03:
I understand what you are saying and I see - it is counter intuitive, but I also think it is redundant. 

I just tried it as explained.

I selected first from the Origin, the XY-plane, then did Attach Sketch... and said OK to the sketch shown, and then I get to the Sketch Attachment dialog and it suggests Plane face, but I had already selected the XY-plane and while Object's XY was in the choices, the intuitive thing would be for Object's XY to be the suggested. But, either way, Plane face or Object's XY :: both result in the XY-plane which I had to select at the beginning of the whole chain of commands to be what the actual attachment is - XY-Plane (XY-plane)

I think the second dialog should have suggested Object's XY or eliminate the second dialog since I already had to select something before I did Attach Sketch...  I tryed it by making a second sketch and Padding it and then selecting a face on that to Attach Sketch... and I still think the second dialog is redundant. 

What makes sense to me is if a command requires item A and item B so that the operation C can be applied then those should be the able to be selected (in the correct order of course) and then the operation (for a different example - Additive Loft).

So maybe what we have is style of operations inconsistancies that I am stumbling with... 

I tried selecting the XY-plane then the Sketch and then Attach Sketch... but I only get Interia 2-3 as the suggested, no Plane face or Object's XY and if I pick that, I get three errors:
21:19:57  PositionBySupport: AttachEngine::getInertialPropsOfShape: infinite shape provided
21:19:57  24373.9 <App> Document.cpp(2540): The graph must be a DAG.
21:19:57  Sketch: AttachEngine::getInertialPropsOfShape: infinite shape provided

mwganson, 2024-11-03:
Make sure in the attachment dialog that the sketch is not one of the references selected for you at the top.  Sometimes you must click the button to enable selection mode for that reference.  I often make this mistake myself and get a message that the desired attachment mode won't work with my selections, and then I look up and there are references that I didn't want in there because I had them selected before or when opening the dialog or creating the sketch.

maxwxyz, 2024-11-03:
Is there an actual issue now?

Buffalo1John, 2024-11-03:
It's not a bug. 

It could be implemented so that it is more consistant with other commands, but that could be said of other operations. The operation could be made to not require a redundant selection which was required to start the operation. 

I think it is a "feature request".

closed 2024-11-05 by yorikvanhavre