FreeCAD/FreeCAD #17633: Command Pallet

jpedrick, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use Visual Studio Code for my daily work. One feature I find indispensable is the command pallet. I just type `Command + Shift + P` and I can search all the possible commands and settings by typing some search terms. The matching commands are displayed along with their keyboard shortcuts on the righthand side. Ideally, if a command has an associated button it would also show the button icon next to the keyboard shortcut.

I also spend quite a lot of time using FreeCAD, but it's been a few years. Now that I'm coming back to it I have forgotten what all the buttons mean and what menus the commands I need are in. The command pallet would make FreeCAD *so much more accessible* and shouldn't be too difficult to implement from a software development perspective, just build a search index from the existing menus and button tooltips. 

### Full version info

```shell
0.21.2-33771 on MacOS M1
```

### Subproject(s) affected?

None

### Anything else?

Please try out the command pallet in Visual Studio Code if you haven't already!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaakmal, 2024-11-03:
This is more likely to be added if you add the label Type:Feature

There are already two partial search functions you can use in the meantime: one under "Help" in the Apple toolbar and autocomplete in the Python Console (accessed by typing <Class>. ).

depthoffocus, 2024-11-03:
There is an extension that does this:

https://github.com/ddfisher/FreeCAD-CommandPalette

maxwxyz, 2024-11-03:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/8290

closed 2024-11-03 by maxwxyz