FreeCAD/FreeCAD #17634: Crash on attempting to open this file.

Stokestack, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you double-click on the attached file, FreeCAD crashes.

Crash report also attached.
[FreeCAD_crash.txt](https://github.com/user-attachments/files/17609452/FreeCAD_crash.txt)
[FlumeBattery_redo.FCStd.zip](https://github.com/user-attachments/files/17609453/FlumeBattery_redo.FCStd.zip)

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-03:
This is a duplicate of the fixed issue https://github.com/FreeCAD/FreeCAD/issues/17280

Current workaround is to either use a recent weekly or open the program before opening a document

closed 2024-11-03 by luzpaz

Stokestack, 2024-11-03:
Thanks! I'll get a later build.