FreeCAD/FreeCAD #17636: Flexible/rigid subassemblies: tooltip is misleading

furgo16, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After merging https://github.com/FreeCAD/FreeCAD/pull/15629 subassemblies can be added as rigid or flexible. The tooltip when adding a component states:

> If checked, the inserted sub-assemblies will not be flexible.
> Rigid means that the sub-assembly will be considered as a solid.
> Flexible means that the sub-assembly joints will be taken into account in the main assembly.
> **You can change this property of sub-assemblies at any time by right clicking them.**

[image]

Unlike stated on the tooltip, right-clicking sub-assembiles does not give the ability to change their rigidness. Or at least I've not found a way to do it.

Instead, the **Rigid** property can be changed on the Property Editor

[image]

I believe the tooltip should read

```diff
- You can change this property of sub-assemblies at any time by right clicking them.
+ You can change this behavior at time by editing the Rigid property of a given sub-assembly in the Property Editor.
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git) Snap 1179
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-11-05:
[image]
Right click appears to work on my side.
It is a right click on the tree view object item.

furgo16, 2024-11-16:
It is indeed there. I do not know why I did not see it the first time, probably a user error on my part. I'll leave this issue open for now, as the related PR adds some more context to the tooltip.

PR #17637 (2024-11-16): Reword rigid/flexible sub-assemblies tooltip.
Fixes #17636
commit 02847d25 (2024-11-16): Reword rigid/flexible sub-assembiles tooltip. Fixes: #17636
closed 2024-11-16 by maxwxyz