FreeCAD/FreeCAD #17638: Sketcher: Crash when trying to change origin

CarstenH5, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please take a look into the video.
FC crashes to desktop.

https://github.com/user-attachments/assets/f7e2c78a-679d-4c42-9062-5dca3450ee72

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-03:
Cannot reproduce on Windows
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

CarstenH5, 2025-08-25:
@luzpaz 

it works on this version i downloaded and build:

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.43049 (Git)
Build date: 2025/08/25 09:37:30
Build type: Unknown
Branch: main
Hash: 4d8a277926937cffd50aa167ca74e6c8ad2d0e49
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * ThreadProfile 1.96.0

Also i can't reproduce the error on this version:

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.38641 +678 (Git) Snap 1634
Build type: Release
Branch: tag: 1.0.2
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, OCC 7.7.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.64

So it's solved i guess :-)

closed 2025-08-25 by maxwxyz