FreeCAD/FreeCAD #17641: Equal sketches in multisection Additive Pipe result in ignored intermediate sections

DonBrus, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design, when performing a multisection Additive Pipe, if the start and end sections coincide (the former being the sweeped, and the latter being added amongst the multisection list), the resulting solid will ignore the added sections and effectively result in a constant pipe. 

In the attached file, try changing the vertical dimension of *end_section* from __4.2__ to __4.0__ to reproduce.
[additive_pipe_issue.zip](https://github.com/user-attachments/files/17610146/additive_pipe_issue.zip)

No errors are generated as result of this issue.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-03:
Sounds like the fixed https://github.com/FreeCAD/FreeCAD/issues/5855 but for additive pipe instead of loft.
@AIRCAP FYI

AIRCAP, 2024-11-17:
This is a different issue -- #5855 was a problem when the sektions were actually in the same location.

Here they are not. they simply have the same shape. However that also isn't sufficient to trigger the issue - if you set both the start section and the end section to 4,2 mm there is no problem - the problem only occurs if they are set both to 4.0

AIRCAP, 2024-11-17:
setting ALL sections to 4,2 mm also triggers the issue - so the issue really happens in this particular case if all the sections have the same height at the border

AIRCAP, 2024-11-17:
I think that's some obscure OCCT bug. not much we can do

AIRCAP, 2024-11-17:
btw this also happens in latest main - issue confirmed