FreeCAD/FreeCAD #17644: SIGSEGV - OpenURL from BIM Workbench Welcome popup

mdavidsen, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Application crashes consecuently when trying to open links from the "Welcome" screen.

to reproduce (fresh user/install): 
* Install flatpak RC version (as described here: https://wiki.freecad.org/Flatpak)
* Open application
* Select BIM workbench
* The "Welcome" popup appears, click either the "BIM tutorial" or "complete documentation" links
* Observe that the application crashes (see stacktrace & screencapture)

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 3d63fc6)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

Screencapture:

https://github.com/user-attachments/assets/63821dda-6ea0-40e8-b034-0000adfa36a8

Stacktrace:

```
flatpak run --branch=beta org.freecad.FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Migrating Start Workbench to Start command... done.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3ee80) [0x7fd234051e80]
#1  0x7fd22f0610e8 in QtWaylandClient::QWaylandWindow::requestXdgActivationToken(unsigned int) from /usr/lib/x86_64-linux-gnu/libQt6WaylandClient.so.6+0x8
#2  0x7fd235537da3 in QGenericUnixServices::openUrl(QUrl const&) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x213
#3  0x7fd235355f22 in QDesktopServices::openUrl(QUrl const&) from /usr/lib/x86_64-linux-gnu/libQt6Gui.so.6+0x2e2
#4  /app/lib/python3.11/site-packages/PySide6/QtGui.cpython-311-x86_64-linux-gnu.so(+0x157034) [0x7fd1ea757034]
#5  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x22271f) [0x7fd23622271f]
#6  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyObject_Vectorcall+0x38) [0x7fd2361c7938]
#7  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x2db6) [0x7fd236106196]
#8  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2176) [0x7fd2362d2176]
#9  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1d441e) [0x7fd2361d441e]
#10  0x7fd23984e2b2 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /app/lib/libpyside6.cpython-311-x86_64-linux-gnu.so.6.7+0x52e
#11  /app/lib/libpyside6.cpython-311-x86_64-linux-gnu.so.6.7(+0x20ab8) [0x7fd239851ab8]
#12  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x1e597f) [0x7fd2347e597f]
#13  0x7fd235b27e19 in QLabel::linkActivated(QString const&) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x39
#14  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x1e55fe) [0x7fd2347e55fe]
#15  0x7fd235bf7d0c in QWidgetTextControl::linkActivated(QString const&) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x3c
#16  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x3ff313) [0x7fd235bff313]
#17  /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6(+0x3ffdab) [0x7fd235bffdab]
#18  0x7fd235c067e6 in QWidgetTextControl::processEvent(QEvent*, QTransform const&, QWidget*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x2a6
#19  0x7fd235bfa802 in QWidgetTextControl::processEvent(QEvent*, QPointF const&, QWidget*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x82
#20  0x7fd235b2b870 in QLabel::keyPressEvent(QKeyEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0xa0
[... 35 lines cut ...]
#56  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x2db6) [0x7fd236106196]
#57  /usr/lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d2176) [0x7fd2362d2176]
#58  /app/lib/python3.11/site-packages/PySide6/QtCore.cpython-311-x86_64-linux-gnu.so(+0x2cdc29) [0x7fd1ea2cdc29]
#59  0x7fd2347d7f08 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x258
#60  0x7fd23599e3d8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Widgets.so.6+0x88
#61  0x7fd2379365c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x98
#62  0x7fd234782e40 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x140
#63  0x7fd234786d75 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x3f5
#64  /usr/lib/x86_64-linux-gnu/libQt6Core.so.6(+0x47517f) [0x7fd234a7517f]
#65  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x239) [0x7fd232dcfe59]
#66  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbe2b8) [0x7fd232e2d2b8]
#67  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fd232dcd513]
#68  0x7fd234a7491f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x6f
#69  0x7fd23478fa4b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x26b
#70  0x7fd23478bc4d in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt6Core.so.6+0x9d
#71  0x7fd23785bd9e in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0xcfe
#72  FreeCAD(+0x95f7) [0x558463ab05f7]
#73  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2808a) [0x7fd23403b08a]
#74  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7fd23403b14b]
#75  FreeCAD(+0x9945) [0x558463ab0945]
````

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hasecilu, 2024-11-03:
Same on self-compiled version with Qt6 on Arch Linux. Which OS are you using?

maxwxyz, 2024-11-03:
@yorikvanhavre FYI

mdavidsen, 2024-11-03:
> Same on self-compiled version with Qt6 on Arch Linux. Which OS are you using?
I'm on Fedora 40

````
uname -a
Linux host 6.11.4-201.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Oct 20 15:04:22 UTC 2024 x86_64 GNU/Linux
```

yorikvanhavre, 2024-11-04:
I can't reproduce this (Debian testing/qt5). I'll try with Qt6...

> The UI file was made with Qt Designer 5 and we are using Qt6, could be that the problem?

Normally not. Does this cause a crash too?

```python
import QtGui
QtGui.QDesktopServices.openUrl("https://google.com")
```

hasecilu, 2024-11-04:
> Normally not. Does this cause a crash too?
> 
> ```python
> from PySide import QtGui
> QtGui.QDesktopServices.openUrl("https://google.com")
> ```

Works fine

It's wayland stuff, it works fine on Openbox.

---

```
OS: LainOS (XFCE/openbox/xcb)
Architecture: x86_64
Version: 1.1.0dev.36460 +2653 (Git)
Build type: Release
Branch: test/prs
Hash: 440b96eaf3a2a12a9079b3944101a11cb53a25fe
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/

```

closed 2025-09-22 by github-actions[bot]