FreeCAD/FreeCAD #17646: Core folders require reorganization

davesrocketshop, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Core folders contain a large number of unorganized files
    - src/Gui: 582
    - src/App: 195
    - src/Base: 171
- Searching for specific files is time consuming
- Working on code touching multiple folders leads to excessive scrolling
- Time to organize with subfolders?

Perhaps this makes sense to do during the c++20 conversion and use of packages

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windows11
Installed mods: 
  * ArchTextures
  * CfdOF 1.27.13
  * fasteners 0.5.29
  * fcgear 1.0.0
  * FeedsAndSpeeds 0.5.0
  * GDML.backup1701064072.5104816 2.0.0 (Disabled)
  * kicadStepUpMod 11.3.9
  * LCInterlocking
  * MaterialTest
  * OpenTheme 2024.9.1
  * render 2024.6.11
  * Rocket 4.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-04:
ping @wwmayer would you mind weighing-in ?

wwmayer, 2024-11-04:
Sure, we can create further sub-directories and move files there. But  this should also be done for some application modules like PartDesign, not only the Core system.

davesrocketshop, 2024-11-04:
I think a key question would be when to do it. It will be disruptive.

As I mentioned one possibility would be when we add the C++20 support, but that won't be until the April retirement of Ubuntu 20.04 support. Should this be started sooner since we're at the start of the next dev cycle?

wwmayer, 2024-11-06:
> I think a key question would be when to do it. It will be disruptive.

Not necessarily. We can do it the same way as Qt. For all public classes it provides convenience headers that include the actual header with the class declaration. This way it can be avoided to modify all files that use these headers.

> Should this be started sooner since we're at the start of the next dev cycle?

I don't have a strong opinion on this. This should be discussed in the forum.

wwmayer, 2024-11-06:
Public Qt headers are not in the src directory but in the include directory.

FEA-eng, 2026-02-13:
> Some reorganization has been done, but there is still some left IMHO

I assume it's still relevant then.

FEA-eng, 2026-08-16:
See [this](https://github.com/FreeCAD/FreeCAD/issues/17646#issuecomment-3895260806)

tritao, 2026-08-16:
Looking through the remaining `src/Gui` files, I think there is still a fairly clear set of work before calling this done:

* Promote the existing CMake groups to actual folders: `Commands`, `ViewProviders`, `View3D`, `Editor`, `Workbench`, `Views`, `DockWindows`, `Help`, `Widgets`, and `Parameters`.
* Finish the partially organized areas: move the remaining Inventor files into `Inventor/`, dialog files into `Dialogs/`, `Task*` files into `TaskView/`, and normalize the remaining settings pages under `PreferencePages/`.
* Group the document lifecycle code (`Document*`, observers/model, autosave/recovery/merge) under a `Documents/` folder.
* Do one final pass over the miscellaneous files, moving things like `Expression*` and `InputHint*` where they have a clear owner, while leaving the actual application-level infrastructure at the root.

I’d tackle these as several focused PRs. Once that is done and the remaining root files are there intentionally, I think `src/Gui` can be considered finished for this issue.