FreeCAD/FreeCAD #17648: Fails to add secondary projections

trilbytim, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Yesterday, Tech Draw was working OK, today it won't add Secondary Projections, I'm not aware of anything I've done to cause this, but the the issue is affecting both my build from source very latest version (just pulled and rebuild this evening) AND an older weekly build running from an AppImage, so it must be something very specific to my system. I did change some values in Preferences, but as far as I can remember, I have set these back to standard. Happy to add any other files to help find how the system got in this state (where are the 'Preferences' stored on the system?)
### Problem:
Attempting to add a secondary projection to an existing view (of any form) gives the error message below when ticking the box to generate it:
`pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: {'sclassname': 'N4Base12RuntimeErrorE', 'sErrMsg': 'Cannot get value from invalid enumeration', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Cannot get value from invalid enumeration', 'btranslatable': False, 'breported': False}
19:01:56  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: {'sclassname': 'N4Base12RuntimeErrorE', 'sErrMsg': 'Cannot get value from invalid enumeration', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Cannot get value from invalid enumeration', 'btranslatable': False, 'breported': False}`
[image]
A "ProjGroup" item is created in the model tree, but nothing appears on the drawing and the item doesn't appear to have any data associated with it:

I've added a very simple file demonstrating the problem. 
[Tech Draw bug.zip](https://github.com/user-attachments/files/17611744/Tech.Draw.bug.zip)
[image]

### Full version info

```shell
Problem exists with BOTH these versions today, but didn't exist with EITHER version yesterday!

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39124 (Git)
Build type: Unknown
Branch: main
Hash: f21ae1fa9dcc62d4d7cbfe1b1ddf0fea7641b5a8
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * sheetmetal 0.5.3
  * OpenTheme 2024.9.1
  * Plot 2024.10.23
  * fasteners 0.5.29

AND:
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * sheetmetal 0.5.3
  * OpenTheme 2024.9.1
  * Plot 2024.10.23
  * fasteners 0.5.29
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-11-03:
Your projection group has no source.   If asked, I would have said this is impossible to do without python.  
Do you remember the steps you followed to create the projection group?

[image]

WandererFan, 2024-11-03:
Your preferences should be in ~/.config/FreeCAD/user.cfg

trilbytim, 2024-11-04:
@WandererFan I was literally doing what was shown in the first screenshot. In the Task Dialog for View I was clicking the tick box for 'Secondary Projection' and was getting the error message that I shared and it was creating a projection group with no source.

### UPDATE:
I have fixed this now by renaming the file .config/FreeCAD/user.cfg to force it to create a new one when I restarted FC. It now behaves as it should, creating secondary projections linked to the first view. So there was something in my user.cfg file that was causing this problem. I have attached the contents of my .config/FreeCAD directory below. I had been clicking settings in Edit>Preferences>TechDraw before this problem happened, so that was probably what caused it, but as far as I could tell I had set all those preferences back to standard and the problem persisted until I did the 'nuclear option' just now of starting with a fresh config file.
[Tech Draw bug.zip](https://github.com/user-attachments/files/17616108/Tech.Draw.bug.zip)

WandererFan, 2024-12-22:
If there is no source shape there will be no secondary views as there is nothing to draw.
Should probably be a message from Insert View that there is nothing to draw.

closed 2024-12-22 by WandererFan