FreeCAD/FreeCAD #17650: FEM : Cannot assign the material to the part

leonehard, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I double click on MaterialSolid the material assignment and editor page does not open.

### Full version info

```shell
C:/Program Files/FreeCAD 1.0/data/Mod/Material/Resources/Materials/Standard/Metal/Aluminum/AlMg3F24.FCMat
21:46:44  
21:46:44  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femviewprovider\view_material_common.py", line 58, in setEdit
    super().setEdit(vobj, mode, task_material_common._TaskPanel)
  File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femviewprovider\view_base_femobject.py", line 89, in setEdit
    task = TaskPanel(vobj.Object)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femtaskpanels\task_material_common.py", line 182, in __init__
    self.selectionWidget = selection_widgets.GeometryElementsSelection(
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femguiutils\selection_widgets.py", line 235, in __init__
    self.initUI()
  File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femguiutils\selection_widgets.py", line 275, in initUI
    ).format("<br>", "<b>", self.sel_elem_text, "</b>")
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'IndexError'>: Replacement index 4 out of range for positional args tuple
```

### Subproject(s) affected?

FEM

### Anything else?

OK

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-04:
Please edit the above ticket and add your Full [About](https://wiki.freecad.org/About) info. thank you

luzpaz, 2024-11-04:
Looks similar to #17561

FEA-eng, 2024-11-04:
> Looks similar to #17561

That one is just a limitation. The one here would be a serious bug if confirmed but we need version info.

leonehard, 2024-11-04:
In FreeCAD 1.0.0RC2 I click in (A) and after When I click in create material ( yellow ball ) don't open the assign material page. 
Sorry for my english.

FEA-eng, 2024-11-04:
Isn’t your FreeCAD language set to Italian by any chance? Can you try in the newest weekly build?
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-08-25:
No clear response from @leonehard on multiple tries. Closing.