FreeCAD/FreeCAD #17651: Sketcher: Select multiple lines, can't apply any constraints

theosib, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[shoe cabinet.FCStd.zip](https://github.com/user-attachments/files/17612088/shoe.cabinet.FCStd.zip)
[image]

Steps to reproduce:

1. Open Sketch
2. Select all of the horizontal lines except for the top and bottom ones
3. Press L or otherwise select a distance or length constraint tool

Expected result:
A dialog should pop up, offering me the ability to enter a length.

Actual result:
Nothing happens.

Notes:
If I select only one line, I can set a constraint. If I select multiple, I cannot. This is a regression. I didn't have this problem in 38946.

### Full version info

```shell
OS: macOS 15.0.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-11-03:
Oh yeah. If you have 100 segments, you'll need to click on each of them to assign dimension constraints.
Although in some cases it's enough to assign equality constraints and set the length of one segment from this equality.

But this method is not suitable for all tasks.

theosib, 2024-11-03:
I know I've done this before, though. I also do this all the time with circle radius/diameter. Select multiple objects, and you can set the constraint on all of them.

AIRCAP, 2024-11-04:
I just tested that in 0.21.2 - i can confirm you can select multiple circles and assign a constraint - this will automatically assign a radius/diameter constraint to the first and equality constraints to all others.

with lines this does not work in 0.21.2 - you get an error message that says "select exactly one line or up to two points from the sketch" for either length, horizontal length or vertical length.

in which version is this supposed to have worked? it doesn't look like a regression.

hyarion, 2024-11-04:
I don't have 38946 but tested it on 39024 and 38923 and it works the same for both of them (both macOS weeklies).
[image]

So I don't think this is a regression, but it is a good feature request.

@theosib for now, the workaround is:
1. select all horizontal lines you want to add the constraint to
2. press E, this creates equals constraints
3. reselect one of the horizontal lines
4. press L

kaiwas, 2024-11-04:
> . I also do this all the time with circle radius/diameter.

To say that I am shocked is to say nothing.
Seeing that dimensional constraints are not assigned to straight line segments (with multiple selection) - I did not even try to do this for circles.

I have never seen a more disorganized and uncoordinated program. In almost every "action" we can have different behavior. All this resembles walking on a field with rakes scattered on it.