FreeCAD/FreeCAD #17652: Set units to US customary, everything still display in millimeters

theosib, 2024-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried setting FreeCAD to "US customary" units, but it seems to have no effect at all. The only places I've tested this are in Sketcher and Part Extrude. I would expect changing the units to change what units are displayed and what the units default to when entering new values, but that doesn't happen.

[shoe cabinet.FCStd.zip](https://github.com/user-attachments/files/17612119/shoe.cabinet.FCStd.zip)

### Full version info

```shell
OS: macOS 15.0.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-04:
Since V1.0 units are stored in the document.
https://wiki.freecad.org/Preferences_Editor#General_2

Closing this.

closed 2024-11-04 by Roy-043

theosib, 2024-11-04:
This is all fine, except that I can't find any option to change the units for an existing document. I have tried googling this, and I'm not finding anything other than out-dated explanations for how to change the units in the Preferences menu. So if you've made this save in the document, but there's no way to change the document, then that's a regression.

Roy-043, 2024-11-04:
You can select the document and then edit the "Unit System" property via the Property editor. Or you can use the [Std_ProjectInfo](https://wiki.freecad.org/Std_ProjectInfo) command.

theosib, 2024-11-04:
Aha! I see that now. And also there's an option in the lower right of the window. Thanks!