FreeCAD/FreeCAD #17656: Sketcher: Zoom: wrong zoom + resizing of text in constraints.

AlbertoGonzalezOlmos, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I started yesterday using this version of the software. The sketcher was worked well at the beginning and at some point I couldn't zoom in and out normally. Also, the text of the constraints is resized wrongly. I have attached a screen recording of the issue.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[Screencast from 2024-11-04 08-03-41.webm](https://github.com/user-attachments/assets/de8a6b0b-4324-40fb-980d-4d4357036608)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-11-04:
I cannot reproduce this

maxwxyz, 2024-11-04:
I'll close this if it works in the latest dev version. Comment if there still is an issue

closed 2024-11-04 by maxwxyz

cloudlesssmarthome, 2025-07-05:
still an issue with 1.0.1 MacOS