FreeCAD/FreeCAD #17658: FreeCAD does not gain control of Mac menu when launched

Stokestack, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When FreeCAD is launched, the Mac menu bar continues to show the menu for the previously-focused application.

FreeCAD does respond to input, although performance is extremely poor (which may be unrelated). Upon clicking the name of whatever app appears in the menu bar, it switches to FreeCAD. Screen grab attached.

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/b6fded2e-1986-4280-aab7-f1f8ffa92f1c

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-04:
@JohnOCFII can you confirm this?

nicolasgalvez, 2024-11-04:
I noticed this when I first opened the application after installing on my M2 Mac, but I can't reproduce the issue now. Except for the slow performance.

Stokestack, 2024-11-04:
I saw it gain control of the menu once in several launches today, but the vast majority of the time it doesn't.

Performance report here: https://github.com/FreeCAD/FreeCAD/issues/17659

JohnOCFII, 2024-11-05:
> @JohnOCFII can you confirm this?

I'm not experiencing this issue. Note this is the latest build, fresh as of a couple of hours ago.  I also did not experience this issue with 39092, but didn't have 39100.

```
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39127 (Git)
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```

Stokestack, 2024-11-05:
Still experiencing this with the build 39127, but... it does appear to be related to the Start-page problem

As soon as you close the Start-page tab, FreeCAD gets control of the menu.

hyarion, 2024-11-10:
@Stokestack can you test if this is still an issue in the latest build?

Stokestack, 2024-11-10:
Appears to be fixed! Thanks!

closed 2024-11-11 by maxwxyz