FreeCAD/FreeCAD #17659: Performance is unusable

Stokestack, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The UI responds very slowly in most cases, sometimes taking seconds to respond to mouse clicks or moves. Tumbling on the canvas doesn't appear to work at all.

Screen grab attached.

https://github.com/user-attachments/assets/d67191a1-c4eb-4fff-b521-041943e380ab

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Stokestack, 2024-11-04:
@benj5378 Thanks for the follow-up. In previous versions I haven't had this problem at all; performance has been snappy.

I also did not encounter [this menu problem](https://github.com/FreeCAD/FreeCAD/issues/17658) before; I don't know if they're related.

FreeCAD stands steadily at 80% CPU when idling in the background, so something is definitely up.

oursland, 2024-11-04:
I wonder if PR https://github.com/FreeCAD/FreeCAD/pull/17565 is related.

Stokestack, 2024-11-04:
I'll bet it is. If I close the Start tab, performance is fine.

Stokestack, 2024-11-05:
This does appear to be causing https://github.com/FreeCAD/FreeCAD/issues/17658

If you close the Start-page tab, FreeCAD gets control of the menu.

theosib, 2024-11-05:
I can confirm that the Start screen is causing the 100% CPU usage. If I close the Start tab, the CPU usage problem goes away.

RWL-69, 2024-11-06:
Not sure if it is related but could be.  In Linux, I just reported a bug with high CPU usage with the start page open.  Closing the start page, usage dropped to less than 5%.  I ran an strace and saw a repeated call to open usr/share/examples/Schenkel.stp.  This is the only call in the strace.

This is the issue I just created.

https://github.com/FreeCAD/FreeCAD/issues/17698

hyarion, 2024-11-06:
@maxwxyz this is a duplicate of the closed issue: https://github.com/FreeCAD/FreeCAD/issues/17698 which was fixed by a merged pr today

closed 2024-11-07 by maxwxyz