FreeCAD/FreeCAD #17661: BIM: Shape of BuildingPart results in duplicate geometry in ArchView

Roy-043, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If an ArchView is created from a section of a BuildingPart, both its shape and the shape of the objects in its Group are processed. Since the shape of a BuildingPart duplicates the shapes in the Group, every shape is sectioned twice.

1. Open the attached [file](https://github.com/user-attachments/files/17618525/buildingpart-shape-problem.FCStd.zip) (remove the .zip extension).
2. Paste the code below in the Python console.
3. Result: ArchView (from the BP) has 2x 8 paths. ArchView001 (from the Wall) has 2x 4 paths.

Code:
 ```python
print(App.ActiveDocument.ArchView.Symbol)
print(App.ActiveDocument.ArchView001.Symbol)
```

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-04:
@yorikvanhavre FYI

yorikvanhavre, 2024-11-05:
Ok the culprit is the `addgroups` parameter of Draft_get_group_contents...
```
>>> Draft.get_group_contents(s1.Objects)[0].Label
'Wall'
>>> Draft.get_group_contents(s2.Objects)[0].Label
'Wall'
>>> Draft.get_group_contents(s1.Objects, addgroups=True)[0].Label
'Level'
>>> Draft.get_group_contents(s2.Objects, addgroups=True)[0].Label
'Wall'
```

I'm not too sure what to do here... The Shape2Dview does more or less the same but does not set addgroups to True... I can't remember the reason we did it here. I'll test both options

Roy-043, 2025-08-26:
Still an issue.

closed 2026-03-09 by chennes