FreeCAD/FreeCAD #17665: Spacemouse not working on latest weekly builds

mrKeyMojo, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Spacemouse Wireless is not working anymore on the latest weekly builds. No movement at all with spacemouse in any direction. Also no reaction on the side buttons. Last build that it worked on for me was 1.1.0dev 39057. Also tested build 39092 without success. I have not tested versions between 39057 and 39092. Driver version spacemouse: 10.8.5

### Full version info

```shell
OS: macOS 13.6.9
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-04:
No issues on Windows.
@pskowronskiTDx FYI
Maybe related https://github.com/FreeCAD/FreeCAD/pull/17000

rstadie, 2024-11-13:
> No issues on Windows. @pskowronskiTDx FYI Maybe related #17000

Yes, this is related. The mentioned PR fixes this issue.

mrKeyMojo, 2024-12-02:
I think this is solved, the spacemouse works again on latest weekly builds. Latest tested on build 39380.

closed 2024-12-02 by maxwxyz