FreeCAD/FreeCAD #17668: External geometry from other body: Selected external reference plane must be normal to sketch plane

Beep6581, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create body1 with a sketch and some geometry.
2. Create body2.
3. Import some geometry from body1 into body2 using a sub-shape binder.
4. Create a sketch in body2.
5. Try to create external geometry by clicking an edge. It doesn't allow me to select an edge.
6. Try to create external geometry by clicking a face. I get error "Selected external reference plane must be normal to sketch plane".

Same if I pad the sketch in body1 and try to use some feature of it through a sub-shape binder in body2.

I'm new to this, but I could swear I did exactly this just yesterday and it worked.

I may have been using a not-HEAD version yesterday, in which case this might be a regression.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39104 (Git)
Build type: Release
Branch: main
Hash: 815375a4a0d65f9717f6f7b85a449cf76e2987ed
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.29
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.12
  * Curves 0.6.51
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Beep6581, 2024-11-04:
https://github.com/user-attachments/assets/460ec30d-6f52-45da-8a60-5a14b4c5f58a

maxwxyz, 2024-11-04:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13313

closed 2024-11-04 by maxwxyz