FreeCAD/FreeCAD #17669: Add on manager fails in macOS Sonoma

rmgvstd, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The add on manager in FreeCAD fails to load data from GitHub. Running FreeCAD 0.21.2 on an M1 Mac Studio with 64GB RAM and MacOS Sonoma 14.7. This error message is returned:Could not load README data from URL https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/README.md
If I go ahead and try to install then the error message in the screenshot is returned.
I tried to install Add ons manually as recommended in other posts but can't find the Mod directory even after making hidden directories visible. It just isn't in the place it's supposed to be (or any other replace for that matter)

### Full version info

```shell
[code]
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Addon Manager

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-11-04:
@chennes FYI

maxwxyz, 2024-11-04:
Can you please try in the recent weekly build https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds or the RC version?

rmgvstd, 2024-11-04:
Ok. Thanks will do

chennes, 2024-11-04:
This is a known, and fixed, bug. Apple *always* has `git` available as an executable, so it turns out we couldn't rely on that for detection. The workaround is to either disable git in the addon manager preferences, or to use Xcode to install git (basically by installing Xcode).

closed 2024-11-04 by chennes

rmgvstd, 2024-11-05:
Thanks. Once GIT was installed 0.21 was able to load add ons although I still get the message about not being able to get the readme file.RegardsStephen DevereuxSent from my iPad