FreeCAD/FreeCAD #17671: "TechDraw did not retrieve any geometry for View/View" error when trying to create a view of a large part

FEA-eng, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create Part Cube, enter 11 m (not mm) for each dimension
2. Add TechDraw page and try inserting a view

There's an error message:

`TechDraw did not retrieve any geometry for View/View`

until the default 1:1 scale is adjusted to something more reasonable like 1:100. Of course, I'm not going to create a view in the original scale but the error message is really misleading. I thought I forgot to select the object when creating a view.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-04:
Is it the same as in the stable version?

FEA-eng, 2024-11-04:
> Is it the same as in the stable version?

No such error message there, the view just appears as an empty frame until rescaled.

WandererFan, 2024-11-05:
If you set "Allow crazy edges" to true on the Advanced tab, then create a 1:1 view of the 11m cube it will work.  An 11m edge on paper is considered "crazy".

Agree that the message is not helpful in this case.  We could print a warning at the point the crazy edge is detected.  Maybe it is time to retire that crazy edge test.

FEA-eng, 2024-12-05:
> We could print a warning at the point the crazy edge is detected.

That can be a good idea. It would be less confusing if the view was created but there was a warning that something went wrong.

WandererFan, 2024-12-22:
This is fixed. Adding a 10m cube to a page works.

closed 2024-12-22 by WandererFan

ErtugrulEge, 2025-03-24:
I got the TechDraw did not retrieve any geometry for View/View 

[image]

[image]

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

[revolve.zip](https://github.com/user-attachments/files/19416028/revolve.zip)