FreeCAD/FreeCAD #17677: Reliable crash on exporting STEP file

theosib, 2024-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[show cabinet 2.FCStd.zip](https://github.com/user-attachments/files/17624256/show.cabinet.2.FCStd.zip)

I get a crash if I select Part and export it as a STEP file.

### Full version info

```shell
OS: macOS 15.0.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-11-05:
I was able to successfully select the full `Part` export as STEP, and even import and slice within PrusaSlicer.  Note that I used tonight's weekly, which was a bit newer than the version you used.

```
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39127 (Git)
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```
[image]

theosib, 2024-11-05:
I tried 39127. It still crashes for me. It spends about a minute with the FreeCAD UI still responsive (like the export is happening in a background thread), and then it crashes. No STEP file is written out. I'm 99% sure that the attached crash log is from the crash of 39100 on the shoe cabinet.
[39100_crash.txt.zip](https://github.com/user-attachments/files/17626493/39100_crash.txt.zip)

JohnOCFII, 2024-11-05:
Hmm... that one works for me too.

Curious what STEP export options are you using?  This is what I've got set:

[image]

hyarion, 2024-11-05:
This works for me too. STEP export options + crash log would be nice :)
<details><summary>version info</summary>
<p>

```
OS: macOS 14.5
Architecture: arm64
Version: 1.1.0dev.39099 +2 (Git)
Build type: Release
Branch: my_branch
Hash: ba08ac359a8494c1f1481e2358683a21138fec7a
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

</p>
</details>

theosib, 2024-11-05:
Good catch on that. I had the legacy export function enabled as part of an attempt to work around another bug I encountered, and I forgot to revert the setting. Unchecking that fixed the problem.

hyarion, 2024-11-05:
By changing it to legacy export I also can reproduce this issue in `FreeCAD-1.1.0dev-39024`

The behavior I see is that I first get an error message `C++ exception thrown (vector)` and then when I try to interact with the ui it will crash.

<details><summary>Stack trace</summary>
<p>

```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000046
Exception Codes:       0x0000000000000001, 0x0000000000000046

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1221]

VM Region Info: 0x46 is not in any region.  Bytes before following region: 4377132986
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104e5c000-104e7c000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD-1.1.0dev-39024.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libpython3.11.dylib           	       0x1059c46c4 _PyFrame_Clear + 344
1   libpython3.11.dylib           	       0x10599ed68 _PyEval_Vector + 208
2   libpython3.11.dylib           	       0x10589c150 method_vectorcall + 344
3   libFreeCADGui.dylib           	       0x1078054e0 Gui::PythonCommand::isActive() + 304
4   libFreeCADGui.dylib           	       0x1077fde08 Gui::Command::testActive() + 544
5   libFreeCADGui.dylib           	       0x10780a124 Gui::CommandManager::testActive() + 60
6   libFreeCADGui.dylib           	       0x107bf77f4 Gui::MainWindow::_updateActions() + 788
7   libQt5Core.5.15.13.dylib      	       0x1072d1e84 0x10718c000 + 1334916
8   libQt5Core.5.15.13.dylib      	       0x1072b5094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
9   CoreFoundation                	       0x18a4ae46c __CFRunLoopDoSource0 + 176
10  CoreFoundation                	       0x18a4ae1dc __CFRunLoopDoSources0 + 244
11  CoreFoundation                	       0x18a4acdc8 __CFRunLoopRun + 828
12  CoreFoundation                	       0x18a4ac434 CFRunLoopRunSpecific + 608
13  HIToolbox                     	       0x194c5019c RunCurrentEventLoopInMode + 292
14  HIToolbox                     	       0x194c4ffd8 ReceiveNextEventCommon + 648
15  HIToolbox                     	       0x194c4fd30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
16  AppKit                        	       0x18dd0bd68 _DPSNextEvent + 660
17  AppKit                        	       0x18e501808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
18  AppKit                        	       0x18dcff09c -[NSApplication run] + 476
19  libqcocoa.dylib               	       0x10b9a44d0 0x10b978000 + 181456
20  freecad                       	       0x104e63184 main + 5128
21  dyld                          	       0x18a0460e0 start + 2360
```

</p>
</details>

theosib, 2024-11-06:
Now, here's a project that crashes the non-legacy STEP export. Export Part004. Crashed every time.
[brackets3.FCStd.zip](https://github.com/user-attachments/files/17640184/brackets3.FCStd.zip)

wwmayer, 2024-11-06:
Do you have set the option to export invisible objects? For me it pops up an error dialog with the message: `NCollection_Array1::Create`
But this is an error we probably can't do much because it happens inside OCC's `STEPCAFControl_Writer::Transfer`.

The other error with the legacy STEP export is a bug in FreeCAD code that can be fixed.

theosib, 2024-11-06:
I did have "Export invisible objects" checked. Unchecking it prevented the crash. Thanks.

wwmayer, 2024-11-06:
@hyarion if you load [brackets3.FCStd.zip](https://github.com/user-attachments/files/17640184/brackets3.FCStd.zip), use the new exporter and set the option to export invisible objects do you also get a crash? And if yes what's the call stack?

I don't see why it crashes for theosib because an exception neither jumps out of a Qt slot function nor an event handler.

hyarion, 2024-11-07:
@wwmayer sure!

Using these options:

[image]

<details><summary>In 1.1.0dev.39024</summary>
<p>

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

</p>
</details> 

I get the following:
<details><summary>stack trace</summary>
<p>

```
Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Codes:       0x0000000000000001, 0x0000000000000018

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [35778]

VM Region Info: 0x18 is not in any region.  Bytes before following region: 4344430568
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102f2c000-102f4c000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD-1.1.0dev-39024.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libTKXDESTEP.7.7.2.dylib      	       0x34234798c 0x3422f4000 + 342412
1   libTKXDESTEP.7.7.2.dylib      	       0x342338d48 STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) + 2284
2   libTKXDESTEP.7.7.2.dylib      	       0x342335cc4 STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) + 3336
3   libTKXDESTEP.7.7.2.dylib      	       0x342334ed8 STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) + 336
4   Import.so                     	       0x34218fdb8 Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const + 140
5   ImportGui.so                  	       0x3420194bc ImportGui::Module::exporter(Py::Tuple const&, Py::Dict const&) + 3444
6   libFreeCADBase.dylib          	       0x1035f8f9c method_keyword_call_handler + 412
7   libpython3.11.dylib           	       0x1039bfe20 cfunction_call + 124
8   libpython3.11.dylib           	       0x1039685e4 _PyObject_MakeTpCall + 332
9   libpython3.11.dylib           	       0x103a6a774 _PyEval_EvalFrameDefault + 46152
10  libpython3.11.dylib           	       0x103a5e35c PyEval_EvalCode + 220
11  libpython3.11.dylib           	       0x103ac44f8 run_mod + 144
12  libpython3.11.dylib           	       0x103ac80a4 PyRun_StringFlags + 112
13  libFreeCADBase.dylib          	       0x10364a5c4 Base::InterpreterSingleton::runString(char const*) + 108
14  libFreeCADGui.dylib           	       0x1058cf948 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 164
15  libFreeCADGui.dylib           	       0x1057bdc30 Gui::Application::exportTo(char const*, char const*, char const*) + 1368
16  libFreeCADGui.dylib           	       0x1058e1cb8 StdCmdExport::activated(int) + 2572
17  libFreeCADGui.dylib           	       0x1058ce6f4 Gui::Command::_invoke(int, bool) + 516
18  libFreeCADGui.dylib           	       0x1058ce4b4 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
19  libQt5Core.5.15.13.dylib      	       0x1053a1e84 0x10525c000 + 1334916
20  libQt5Core.5.15.13.dylib      	       0x105385094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
21  CoreFoundation                	       0x18a4ae46c __CFRunLoopDoSource0 + 176
22  CoreFoundation                	       0x18a4ae1dc __CFRunLoopDoSources0 + 244
23  CoreFoundation                	       0x18a4acdc8 __CFRunLoopRun + 828
24  CoreFoundation                	       0x18a4ac434 CFRunLoopRunSpecific + 608
[... 66 lines cut ...]
    frame #40: 0x0000000194c5019c HIToolbox`RunCurrentEventLoopInMode + 292
    frame #41: 0x0000000194c4fe2c HIToolbox`ReceiveNextEventCommon + 220
    frame #42: 0x0000000194c4fd30 HIToolbox`_BlockUntilNextEventMatchingListInModeWithFilter + 76
    frame #43: 0x000000018dd0bd68 AppKit`_DPSNextEvent + 660
    frame #44: 0x000000018e501808 AppKit`-[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
    frame #45: 0x000000018dcff09c AppKit`-[NSApplication run] + 476
    frame #46: 0x000000010951f758 libqcocoa.dylib`___lldb_unnamed_symbol3677 + 968
    frame #47: 0x000000010994ab88 libQt5Core.5.15.8.dylib`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 312
    frame #48: 0x000000010994d158 libQt5Core.5.15.8.dylib`QCoreApplication::exec() + 120
    frame #49: 0x0000000104a97b7c libFreeCADGui.dylib`(anonymous namespace)::tryRunEventLoop(mainApp=0x000000016fdfe238) at Application.cpp:2200:5
    frame #50: 0x0000000104a86678 libFreeCADGui.dylib`(anonymous namespace)::runEventLoop(mainApp=0x000000016fdfe238) at Application.cpp:2220:9
    frame #51: 0x0000000104a86018 libFreeCADGui.dylib`Gui::Application::runApplication() at Application.cpp:2297:5
    frame #52: 0x000000010000c528 FreeCAD`main(argc=3, argv=0x000000016fdfee78) at MainGui.cpp:295:13

    frame #53: 0x000000018a0460e0 dyld`start + 2360
```

</p>
</details> 

It stops on this line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/WriterStep.cpp#L51:
```c++
    writer.Transfer(hDoc, STEPControl_AsIs);

```

wwmayer, 2024-11-07:
Thanks. So, the question is if this is a real OCC bug or if the input is invalid.

> `writer.Transfer(hDoc, STEPControl_AsIs);`

For me it stops here too but with an exception, not a crash.

So, for now I will re-open the ticket.

PR #17706 (2024-11-06): Import: Fix index error with legacy STEP export
Fixes #17677
commit 41806782 (2024-11-06): Import: Fix index error with legacy STEP export

Fixes #17677