FreeCAD/FreeCAD #17679: Crash trying to deselect dimension tool (seg fault, MacOS)

mechanical-dance, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've had a few crashes trying to create or modify constraints. I am _very_ new to free cad, but i do have software experience, so i may or may not be the ideal tester for the release candidate lol. I've had other crashes while clicking around, but i'm starting to get my bearings and i haven't been able to reproduce this specific issue 

excerpts from crash log:
```
vmRegionInfo" : "0 is not in any region.  Bytes before following region: 4298948608\n      REGION TYPE                    START - END         [ VSIZE] PRT\/MAX SHRMOD  REGION DETAIL\n      UNUSED SPACE AT START\n--->  \n      __TEXT                      1003cc000-1003ec000    [  128K] r-x\/r-x SM=COW  \/Applications\/FreeCAD.app\/Contents\/Resources\/bin\/freecad",
  "exception" : {"codes":"0x0000000000000001, 0x0000000000000000","rawCodes":[1,0],"type":"EXC_BAD_ACCESS","signal":"SIGSEGV","subtype":"KERN_INVALID_ADDRESS at 0x0000000000000000"},
  "termination" : {"flags":0,"code":11,"namespace":"SIGNAL","indicator":"Segmentation fault: 11","byProc":"exc handler","byPid":1502},
  "ktriageinfo" : "VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter\nVM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter\nVM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter\nVM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter\nVM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter\n",
```
```
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   SketcherGui.so                	       0x17690b298 Sketcher::isLineSegment(Part::Geometry const&) + 12
1   SketcherGui.so                	       0x176aa200c SketcherGui::ViewProviderSketch::generateContextMenu() + 944
2   SketcherGui.so                	       0x176a9dd08 SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 3712
3   libFreeCADGui.dylib           	       0x102cf9360 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 656
```
``
{"app_name":"freecad","timestamp":"2024-11-04 05:13:26.00 -0500","app_version":"","slice_uuid":"1646732d-9d8d-34fd-aa82-05b4766ebeb8","build_version":"","platform":1,"share_with_app_devs":0,"is_first_party":1,"bug_type":"309","os_version":"macOS 15.1 (24B83)","roots_installed":0,"incident_id":"96AAF202-FDE9-449C-895A-8709B446A08B","name":"freecad"}
```
not sure what exactly is relevant for you all, but willing to dig deeper or close if this is a known issue. I saw other reports of segmentation errors but those were mostly on linux

### Full version info

```shell
OS: macOS 15.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mechanical-dance, 2024-11-05:
had a couple more since my last post
```
Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [50553]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4362207216
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104020000-104040000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   SketcherGui.so                	       0x15ec73cac DrawSketchHandlerDimension::mouseMove(Base::Vector2d) + 268
1   SketcherGui.so                	       0x15ed51ec0 SketcherGui::ViewProviderSketch::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) + 1340
2   libFreeCADGui.dylib           	       0x10694d504 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 1076
```

mechanical-dance, 2024-11-05:
a new failure mode trying to attach a valid sketch to a plane

```
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [52862]

Application Specific Information:
abort() called

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19f74e600 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x19f786f70 pthread_kill + 288
2   libsystem_c.dylib             	       0x19f693908 abort + 128
3   libc++abi.dylib               	       0x19f73d44c abort_message + 132
4   libc++abi.dylib               	       0x19f72ba40 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x19f3d13f4 _objc_terminate() + 172
6   libc++abi.dylib               	       0x19f73c710 std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x19f73fcdc __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x19f73fc84 __cxa_throw + 92
9   Measure.so                    	       0x11e37d43c Measure::Measurement::length() const + 792
10  MeasureGui.so                 	       0x122f84038 MeasureGui::QuickMeasure::printResult() + 1480
11  MeasureGui.so                 	       0x122f83428 MeasureGui::QuickMeasure::processSelection() + 52
12  libQt5Core.5.15.13.dylib      	       0x106c81e84 0x106b3c000 + 1334916
13  libQt5Core.5.15.13.dylib      	       0x106c65094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
14  CoreFoundation                	       0x19f86dcc8 __CFRunLoopDoSource0 + 176
15  CoreFoundation                	       0x19f86da2c __CFRunLoopDoSources0 + 244
16  CoreFoundation                	       0x19f86c5c8 __CFRunLoopRun + 840
17  CoreFoundation                	       0x19f86bbc4 CFRunLoopRunSpecific + 588
18  HIToolbox                     	       0x1aacdbf64 RunCurrentEventLoopInMode + 292
```
lmk if this should be a separate issue

maxwxyz, 2024-11-05:
Can you try with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

hyarion, 2024-11-05:
@mechanical-dance we've fixed a lot of mac specific crashes on 1.0RC2 so it would be really good if you could try a weekly build. There's a signing issue currently for the weeklies so you need to run `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app` after copying FreeCAD.app to your application folder to be able to start it.

mechanical-dance, 2024-11-05:
will try the weekly tonight! 

@maxwxyz is there documentation on getting you all an example file/ what your requirements are? i have a bunch of error reports but that’s all. last night i had ~8 crashes with 3 different failure modes, maybe the file i was working on can be used?

hyarion, 2024-11-05:
Any file + step by step instructions of how to reproduce the issue works.

Instructions can be something like this:
1. Load xyz.FCStd
2. Expand Part001
3. Open Sketch092
4. Select top right line
5. Click horizontal constraint
Freecad crashes

Providing a screenshots or video where it is difficult to describe with text is really appreciated.

If you make it easy for us to reproduce the issue we will be able to fix it faster :)

Good luck!

mechanical-dance, 2024-11-06:
Ack. Gonna try the weekly and see if it solves my problems, but if it helps i could play around with the RC this weekend and screen record until it crashes.

hyarion, 2024-11-06:
@mechanical-dance it would actually be better to play around with a weekly. The weekly is like a RC2.5 at this point :)

mechanical-dance, 2024-11-08:
Good to know. I've been using the weekly for the last couple days, no crashes but i've been learning the python script engine so i haven't been clicking as many random things. Will try making something in the part designer this weekend with kbm and if I still get no crashes I'll close the issue. Thanks for the help!

hyarion, 2024-11-13:
@luzpaz sounds like this issue can be closed now.

mechanical-dance, 2024-11-15:
follow up in case anyone ends up here later, after updating to the weekly I only had one crash in about 15-20 hours. after the crash my backups restored me to my exact progress and i've been going for another 5 or so hours with no issues at all. Cheers to the team, and I'm really lucky to have started right after this release!