FreeCAD/FreeCAD #17680: Invisible lines are deleted from sketch along with selected ones

Stokestack, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Non-visible, previously-selected lines are deleted from a sketch when you delete a selected line. This occurs when you need to select a line that's overlapped by another, so you find the overlapping one in the elements list and deactivate it so you can select the overlapped one.

Screen grab attached.

To reproduce, open the attached file and try to delete the small, lower-rightmost line (not the longer one that is coincident). To select only the short segment, click on it and then find the selected line (which should be the long one) in the elements list and uncheck its visibility.
Now select the short line segment and press Delete.
Go back up in the elements list and note that both long and short lines are deleted.

I suppose it's arguable as to whether this is a bug or useful behavior. At the very least I submit that it's unintuitive and likely to create a broken model that the user might not discover until it's impractical to undo.

If it is deemed useful, what is the recommended course of action for a user who faces this selection problem?

Another problem is the continued illegibility of labels in the Tasks panel. Maybe that checkbox says something pertinent, but I can't tell.

https://github.com/user-attachments/assets/efccc70e-2e50-4eb5-b5b8-96fc5687d368

[FlumeBattery_redo.FCStd.zip](https://github.com/user-attachments/files/17626184/FlumeBattery_redo.FCStd.zip)

### Full version info

```shell
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39127 (Git)
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark behave/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-05:
They are still selected. Can you show the output in the selection view?

Stokestack, 2024-11-05:
Thanks for the follow-up. What is the selection view?

maxwxyz, 2024-11-06:
@AjinkyaDahale could you take a look at this?

AjinkyaDahale, 2024-11-18:
The problem appears to be the making the entity invisible doesn't unselect it. Hopefully a quick fix.

Would you prefer the selection to remain even if made invisible and be reflected in the listview, or the invisible object to be unselected?

Stokestack, 2024-11-18:
Thanks for the follow-up. I think the most intuitive course is to deselect it if you make it invisible. I'm trying to consider other cases though; for example, where an object is blocking the view of another but the user still wants to delete both.

An argument could be made for either, but the highlighting of items in that list is not 100% consistent with what is selected, at least as it stands right now. Also, items rapidly scroll out of view... taking their selected status out of view as you hunt for other things you want to select.

If the highlighting in that list were 100% indicative of what's selected, I guess I'd vote for keeping the selection even when the item is invisible... to enable the case I mentioned above.

The middle ground would be to deselect the item when the user makes it invisible, but he could still click on it in the list to re-select it. The checkbox could be seen as toggling both visibility and selection, where's the rest of the row would toggle only selection.

Stokestack, 2025-08-26:
I don't see where anything has been done to address this.

Stokestack, 2025-11-10:
That's great. Thanks guys!