FreeCAD/FreeCAD #17682: Fillet in PartDesign results in a wrong shape for a complex body

YohDeadfall, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem I'm experiencing is a wrong geometry produced by a Fillet operation for a body partly built using the Curves workbench. The front wall which should be flat starts to be three dimensional and has a hole.

[repro.tgz](https://github.com/user-attachments/files/17629845/repro.tgz)

[image]

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.3.12
  * A2plus 0.4.60n
  * Curves 0.6.48
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-05:
You may need to open a ticket in the Curves workbench bugtracker: https://github.com/tomate44/CurvesWB/  
Lets see what @tomate44 thinks?

tomate44, 2024-11-05:
I don't think CurvesWB is faulty.
It is used at the beginning of the process to build the BaseFeature.
To this point GeomCheck reports a valid solid, only with GeomAbs_C0 warning. Full 1e-7 tolerance.
Things start to go wrong at the first fillet.

FEA-eng, 2025-10-25:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```