FreeCAD/FreeCAD #17689: Part: Regression with boolean common

wwmayer, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Boolean intersection (common) used to be a commutative operation, i.e. the order of input shapes has no impact on the result. With the current implementation this is not longer the case. Furthermore, for more than 2 input shapes the result in general is wrong.

Steps:
* Create a part box, cylinder and sphere
* Select the box, cylinder and sphere and do a boolean common
* Select the cylinder, sphere and box and do a boolean common
* Select the sphere, box and cylinder and do a boolean common
* Compare the results: they are all different and they are all wrong

Or in a more formal way:
The result of the input shapes S1, S2, S3, ... Sn used to be S1 ∧ S2 ∧ S3 ... ∧Sn while now it is (S1 ∧ S2) ∨(S1 ∧ S3) ∨ ...∨(S1 ∧ Sn)

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39128 (Git)
Build type: debug
Branch: main
Hash: 4c50f2944287ab966a063ef4091480e65ac99acd
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-05:
Any idea what code change was responsible for regression ?

wwmayer, 2024-11-05:
The way how `TopoShape::makeElementBoolean` handles argument and tool shapes doesn't work with Common. The issue is caused by https://github.com/FreeCAD/FreeCAD/pull/14182

AIRCAP, 2024-11-06:
This also fails when a boolean common is done with a compound.

AIRCAP, 2024-11-06:
I suggest introducing a BooleanSetOperation as a new base class

wwmayer, 2024-11-06:
> At this point it would be better to get rid of the MultiFusion as a separate implementation

That's not possible without breaking many existing project files.

A way to handle this could be to override the Build() method of `FCBRepAlgoAPI_Common`.

kadet1090, 2025-12-19:
So I see it this way:

This issue cannot be blocker - we don't have any example of file that is actually broken by that change, and it seems that we are more closely aligned to OCC right now. Depending on use both behaviors probably make sense and so a group of people will argue that if they are computing common of shape A with B,C,D they want to have A and (B or C or D) - another group of people will want to have common of all of these shapes so A and B and C and D.

My proposal is to update the API to expose enum flag that would allow us to choose wanted behavior. We can then expose this as property of boolean operation for users to choose. For files created before 1.0 we can then set that property to be aligned with older behavior and for newer files we can do current one (to preserve compatibility with 1.0).

DeepSOIC, 2026-01-14:
> Problem here is that we have 1.0 that works with the same behavior and so if we now change it to be like it used to be we risk that we break files created in 1.0.

yes, but a) that's already happened on 0.20->1.0 transition; b) it may be aligned with how occ works, but the way it works now makes no sense. And i highly doubt anyone is relying on this behavior, simply because it's pointless. Also, i hope we don't call every bug a feature now.

If we need an independent multicommon implementation, we can simply clone Part XOR and modify it very slightly to produce a proper multicommon. Or even closer, see Macro Overlap.

DeepSOIC, 2026-01-14:
> we don't have any example of file that is actually broken by that change,

I'm pretty sure i have some... after all, i have bumped into this issue when making something with a purpose. But i couldn't easily find one, so i made my own:

[7-sided-allen-ball-head_uses-multicommon.zip](https://github.com/user-attachments/files/24624674/7-sided-allen-ball-head_uses-multicommon.zip)

[image]