FreeCAD/FreeCAD #17690: Part is moving every click on middle button.

dibor, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 Part is moving every click on middle button.
 
[Video_2024-11-05_153439.zip](https://github.com/user-attachments/files/17633280/Video_2024-11-05_153439.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-05:
Can you provide step-by-step instructions on how to trigger this bug please? TIA

Rexbas, 2024-11-05:
For some navigation styles it's a feature, not a bug.

dibor, 2024-11-05:
> From "luzpaz" ***@***.***>
> To "FreeCAD/FreeCAD" ***@***.***>
> Cc "dibor" ***@***.***>; "Author" ***@***.***>
> Date 05/11/2024 18:22:16
> Subject Re: [FreeCAD/FreeCAD] Navigation: Part is moving every click on
> middle button. (Issue #17690)

Hello.
Very simple.
I did install freecad x64 on Windows x64 system, after install choose 
Russian language for interface and classic theme .
Made small part and when I click middle button on my Logitech mouse part 
is moving every click :(

dibor, 2024-11-06:
Thank U.
This feature of the CAD navigation :)

From "Bas Ruigrok" ***@***.***>

Date 05.11.2024 18:45:56

Roy-043, 2024-11-06:
This is indeed normal for CAD navigation.