FreeCAD/FreeCAD #17692: Pad removes "base" body when sketch doesn't have something already extruded underneath

TheRealOne78, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Padding a sketch that doesn't have a solid underneath it causes it to eliminate the whole "base".

I'm having a hard time coming with a better description/naming, because I don't really know how to describe it, so here's a video:

https://github.com/user-attachments/assets/c7308a58-b672-400b-9f6f-1353ebf29e2a

At the end of the video, when I'm changing the pad's length, the object also gets rendered in a weird way, which I haven't seen before.

### Full version info

```shell
OS: Gentoo Linux (Xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.Unknown
Build type: Release
Python 3.12.7, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * Gridfinity 0.6.0
  * FreeCAD-Gridfinity-Workbench 0.6.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

I had to move the sketch due to #17370 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andor, 2024-11-05:
Not sure if this might be considered as a bug.
In version 0.21 multibody in PartDesign returns explicit error message.
In 1.0 there is a switchbox to enable multibody for part as an experimental feature. If must show error otherwise  and apparently it is not doing that.

TheRealOne78, 2024-11-05:
Well sure.. but the sketch is attached to the same body. There aren't multiple bodies...

Andor, 2024-11-05:
I just checked your file with this version and it works perfectly well:
```
OS: macOS 14.6.1
Architecture: arm64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * woodworking 0.21.2.33771
  * Silk 0.1.5
  * Gridfinity 0.6.0
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * MakerWorkbench 1.0.1
  * Curves 0.6.47
  * Assembly4 0.50.15
```

TheRealOne78, 2024-11-07:
After realizing I could use wine to run the weekly build, (AppImages are broken #17620 ) I can now confirm that it works at least in 39127

closed 2024-11-07 by TheRealOne78