FreeCAD/FreeCAD #17694: Assembly: dragging problem created by fixed joint

maxwxyz, 2024-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum report: https://forum.freecad.org/viewtopic.php?t=91899
---
opening the file Fixed_Joint_test.FCStd with:

[Fixed_Joint_test.zip](https://github.com/user-attachments/files/17637754/Fixed_Joint_test.zip)

1.1.0dev.38728 (Git) AppImage
1.0.0RC2.38806 (Git) AppImage

everything works perfectly:
[image]

after activating the Assembly, dragging the green body with the mouse allows it to be rotated around Revolute001 joint without any problems with smooth movement.

On the other hand, opening Fixed_Joint_test.FCStd with:

1.1.0dev.38898 (Git) AppImage
1.1.0dev.38977 (Git) AppImage
1.1.0dev.38991 (Git) AppImage
1.1.0dev.39057 (Git) AppImage
1.1.0dev.39092 (Git) AppImage

after activating the Assembly, dragging the green body around Revolute001 joint with the mouse makes it impossible to rotate it; however, if you delete the Fixed joint, it becomes possible to rotate the green body around Revolute001 joint without any problems and with smooth movement (obviously, now, the rivet no longer follows the movement of the green body).

### Full version info

```shell
OS: LMDE 6 (faye) (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-05:
I cannot reproduce on
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * CfdOF 1.27.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 4.0.0
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

PaddleStroke, 2024-11-13:
I can reproduce.

luzpaz, 2024-11-13:
@PaddleStroke version info ?

PaddleStroke, 2024-11-13:
Well this is very surprising but after rebasing and rebuilding I cannot reproduce this bug anymore.

The rebase changed from 39176 to 39188 so only 12 commits, and I don't know what could have fixed this. But I can't reproduce anymore now.

 
> OS: Windows 10 build 19045
> Architecture: x86_64
> Version: 1.1.0dev.39188 (Git)
> Build type: Release
> Branch: asm_fixed_move_bug
> Hash: 4bc8cd60e9a9c24ed007f0d3d341ca70498c798a
> Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
> Installed mods: 
>   * A2plus 0.4.60k
>   * Assembly4 0.50.2
>   * Curves 0.6.8
>   * ExtremeProLight 1.5.0
>   * fasteners 0.5.0
>   * FreeCAD - Shortcut.lnk
>   * freecad.gears 1.2.0
>   * FreeCAD_FastenersWB-master 0.4.56
>   * kicadStepUpMod 10.23.0
>   * Ondsel 2024.9.13.02
>   * OpenDark 2023.12.17
>   * ProDarkThemePreferencePack 1.0.0
>   * sheetmetal 0.2.63

adrianinsaval, 2024-11-13:
is this a problem in rc3?

PaddleStroke, 2024-11-18:
This bug is only happening 50% of the tries. Which is why I could not reproduce at some point. Good news is that I finally found root issue

closed 2024-11-18 by yorikvanhavre