FreeCAD/FreeCAD #17698: High CPU usage with just start page open

RWL-69, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Current weekly build and at least a few previous weekly builds of FreeCAD I get high core usage with just the Start page open and not touching the machine.

Close the start page and the usage drops.
In --safe-mode, CPU usage ranged from 82% to 90% on one core.  

Closing the start page, FreeCAD drops to less than 3%

I tested in two previous weekly builds and issue is in one but not the other.

### Full version info

```shell
OS: Fedora Linux 39 (KDE Plasma) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39127 (Git) AppImage
Build type: Release
Branch: main
Hash: 1b1e355b9b8c22ae8ca01c383f6b57cdac8a7c04
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Curves 0.6.50
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * CurvedShapes.backup1713545489.8854659 1.0.5 (Disabled)
  * Defeaturing 1.2.2
  * FEM_FrontISTR 0.1.0
  * slic3r-tools
  * fasteners 0.5.29
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.4.0
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.10.32
  * toSketch 1.0.1
  * freecad.gears 1.3.0
  * LCInterlocking
  * CurvedShapes 1.0.12
  * sheetmetal 0.5.0
  * Manipulator 1.5.7
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Anything else?

Running an strace, there is a call to "/usr/share/examples/Schenkel.stp" that is repeated.  

openat(AT_FDCWD, "/tmp/.mount_FreeCAPjMIhL/usr/share/examples/Schenkel.stp", O_RDONLY|O_CLOEXEC) = 22

It is in this version as well.

OS: Fedora Linux 39 (KDE Plasma) (KDE/plasmax11/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Curves 0.6.50
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * CurvedShapes.backup1713545489.8854659 1.0.5 (Disabled)
  * Defeaturing 1.2.2
  * FEM_FrontISTR 0.1.0
  * slic3r-tools
  * fasteners 0.5.29
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.4.0
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.10.32
  * toSketch 1.0.1
  * freecad.gears 1.3.0
  * LCInterlocking
  * CurvedShapes 1.0.12
  * sheetmetal 0.5.0
  * Manipulator 1.5.7
  * kicadStepUpMod 11.3.9

The issue is not in this version.

OS: Fedora Linux 39 (KDE Plasma) (KDE/plasmax11/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Curves 0.6.50
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * CurvedShapes.backup1713545489.8854659 1.0.5 (Disabled)
  * Defeaturing 1.2.2
  * FEM_FrontISTR 0.1.0
  * slic3r-tools
  * fasteners 0.5.29
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.4.0
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.10.32
  * toSketch 1.0.1
  * freecad.gears 1.3.0
  * LCInterlocking
  * CurvedShapes 1.0.12
  * sheetmetal 0.5.0
  * Manipulator 1.5.7
  * kicadStepUpMod 11.3.9

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

RWL-69, 2024-11-06:
More of the strace

`openat(AT_FDCWD, "/tmp/.mount_FreeCAPjMIhL/usr/share/examples/Schenkel.stp", O_RDONLY|O_CLOEXEC) = 22
fstat(22, {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
fstat(22, {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
lseek(22, 0, SEEK_SET)                  = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
lseek(22, 0, SEEK_SET)                  = 0
lseek(22, 0, SEEK_SET)                  = 0
close(22)                               = 0
stat("/tmp/.mount_FreeCAPjMIhL/usr/share/examples/Schenkel.stp", {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
stat("/tmp/.mount_FreeCAPjMIhL/usr/share/examples/Schenkel.stp", {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
openat(AT_FDCWD, "/tmp/.mount_FreeCAPjMIhL/usr/share/examples/Schenkel.stp", O_RDONLY|O_CLOEXEC) = 22
fstat(22, {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
fstat(22, {st_mode=S_IFREG|0664, st_size=590055, ...}) = 0
read(22, "ISO-10303-21;\r\nHEADER;\r\n/* Schen"..., 16384) = 16384
`

I checked the files in the working and broken AppImages and the file size for Schenkel.stp are the same size and have the same sha1sum.

If I open Schenkel.stp from the start menu, the CPU usage drops to less than 5%.  Close the page, usage increases again.

There is no preview in any of the weekly versions I have tried.

marioalexis84, 2024-11-06:
Confirmed.
It happens on the Start page and if the default theme (FreeCAD Classic) is used.
Changing the theme or switching to another MDI window stops the high CPU usage.

mwganson, 2024-11-06:
Yes, it pegs out one of the cores at 100%.  Ouch.

Related to this, I tried disabling show examples as a test, but that parameter is not being respected by the start page.  I still get the examples even though the parameter (tools -> edit parameters -> preferences -> mod -> start -> ShowExamples = false.

mwganson, 2024-11-06:
It looks like there was a PR to fix this a few days ago.

https://github.com/FreeCAD/FreeCAD/pull/17565

closed 2024-11-06 by maxwxyz