FreeCAD/FreeCAD #17702: v1.0.0RC2, MacOS: crash when double-clicking example file

Lx, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've downloaded this example file from the FreeCAD manual:

https://github.com/yorikvanhavre/FreeCAD-manual/blob/master/files/table.FCStd

When I attempt to open this file with FreeCAD (with FreeCAD initially closed) by double-clicking the file, the FreeCAD splash screen shows for a very brief period then disappears, followed by the Apple "Problem Report" screen:

[image]

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-06:
Should be fixed in latest weekly, please try and report if still an issue https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

closed 2024-11-06 by maxwxyz

Lx, 2024-11-06:
@maxwxyz, unfortunately I cannot test this on the latest weekly as the currently available arm64 build seems to be completely broken:

[image]

I verified that the SHA256 sum is correct, so it's not a download corruption issue.

JohnOCFII, 2024-11-07:
> @maxwxyz, unfortunately I cannot test this on the latest weekly as the currently available arm64 build seems to be completely broken:

There's a signing issue currently for the weeklies so you need to run sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app after copying FreeCAD.app to your application folder to be able to start it.

Lx, 2024-11-07:
Thanks, @JohnOCFII: that got it working.

Can confirm that the reported issue **does not** affect the latest weekly build (1.1.0dev.39127).