FreeCAD/FreeCAD #17704: v1.0.0RC2, MacOS: crash when opening sample IFC file

Lx, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I attempted to open this sample IFC file via the Open dialog:

https://raw.githubusercontent.com/andrewisen/bim-whale-ifc-samples/main/BasicHouse/IFC/BasicHouse.ifc

and FreeCAD segfaulted:

[image]

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Lx, 2024-11-07:
@maxwxyz, tested OK on latest weekly (1.1.0dev.39127).

As the weekly build has a version number of 1.1, does this mean the fix will not make it to the final release of 1.0?

maxwxyz, 2024-11-07:
Will be in 1.0.
Closing this issue as fixed.

closed 2024-11-07 by maxwxyz