FreeCAD/FreeCAD #17708: Part.Face.cut produces an invalid shape on main

je-cook, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On freecad v0.21.2, and I believe v1.0RC2, the below script creates a valid face whereas in the most recent conda version the face is invalid.
The most recent working version of main (I cannot seem to find the version and it no longer exists on the freecad conda repo) was the conda tag: `  - freecad=1.1dev=py311h5536ab1_9`. The last working CI run can be seen here: https://github.com/Fusion-Power-Plant-Framework/bluemira/actions/runs/11443446079.

This script is a minimal stripped down version of one of our tests that is now fails on the assertion. The constituent wires are both closed and valid in both cases.

```python
import freecad
import FreeCAD
import Part
from FreeCAD import Base
import numpy as np

# fmt: off
princetonD_x = np.array([
    4.0, 4.00222799, 4.00891543, 4.02007274, 4.03571724, 4.05587304,
    4.0805709 , 4.10984812, 4.14374827, 4.18232098, 4.22562162, 4.27371098,
    4.32665483, 4.38452351, 4.44739137, 4.51533622, 4.58843867, 4.66678142,
    4.75044842, 4.83952405, 4.93409208, 5.0342347 , 5.14003128, 5.25155722,
    5.36888254, 5.4920705 , 5.62117606, 5.75624426, 5.89730852, 6.04438884,
    6.19748998, 6.35659948, 6.5216857 , 6.69269582, 6.86955379, 7.05215823,
    7.24038049, 7.4340626 , 7.63301538, 7.83701659, 8.0458093 , 8.25910031,
    8.47655885, 8.69781545, 8.92246115, 9.15004697, 9.38008368, 9.61204203,
    9.84535332, 10.07941043, 10.31356926, 10.54715073, 10.77944313, 11.00970512,
    11.23716912, 11.46104518, 11.68052538, 11.89478861, 12.10300579, 12.30434548,
    12.49797972, 12.68309018, 12.85887452, 13.02455282, 13.17937404, 13.32262248,
    13.45362404, 13.57175231, 13.67643434, 13.76715599, 13.84346687, 13.90498464,
    13.9513988 , 13.98247371, 13.99805097, 13.99805097, 13.98247371, 13.9513988,
    13.90498464, 13.84346687, 13.76715599, 13.67643434, 13.57175231, 13.45362404,
    13.32262248, 13.17937404, 13.02455282, 12.85887452, 12.68309018, 12.49797972,
    12.30434548, 12.10300579, 11.89478861, 11.68052538, 11.46104518, 11.23716912,
    11.00970512, 10.77944313, 10.54715073, 10.31356926, 10.07941043, 9.84535332,
    9.61204203, 9.38008368, 9.15004697, 8.92246115, 8.69781545, 8.47655885,
    8.25910031, 8.0458093 , 7.83701659, 7.63301538, 7.4340626 , 7.24038049,
    7.05215823, 6.86955379, 6.69269582, 6.5216857 , 6.35659948, 6.19748998,
    6.04438884, 5.89730852, 5.75624426, 5.62117606, 5.4920705 , 5.36888254,
    5.25155722, 5.14003128, 5.0342347 , 4.93409208, 4.83952405, 4.75044842,
    4.66678142, 4.58843867, 4.51533622, 4.44739137, 4.38452351, 4.32665483,
    4.27371098, 4.22562162, 4.18232098, 4.14374827, 4.10984812, 4.0805709,
    4.05587304, 4.03571724, 4.02007274, 4.00891543, 4.00222799, 4.0
  ])
princetonD_z = np.array([
    -4.84194705, -4.94759094, -5.05316445, -5.15859646, -5.26381438, -5.3687434,
    -5.47330575, -5.57741995, -5.68100008, -5.783955  , -5.88618763, -5.98759416,
   -6.08806333, -6.18747563, -6.28570261, -6.38260607, -6.47803738, -6.57183674,
   -6.66383247, -6.75384038, -6.84166311, -6.92708951, -7.00989417, -7.08983686,
   -7.16666221, -7.24009931, -7.30986154, -7.37564643, -7.43713571, -7.49399542,
   -7.54587626, -7.59241405, -7.6332304 , -7.6679336 , -7.69611969, -7.71737375,
   -7.73127151, -7.73738111, -7.73526515, -7.72448306, -7.70459361, -7.67515783,
   -7.63574207, -7.58592132, -7.52528284, -7.45342988, -7.36998567, -7.27459755,
   -7.16694115, -7.04672474, -6.91369352, -6.76763396, -6.60837797, -6.43580706,
   -6.24985614, -6.05051721, -5.8378426 , -5.6119479 , -5.37301441, -5.12129106,
   -4.85709578, -4.58081629, -4.29291017, -3.99390429, -3.68439353, -3.36503877,
   -3.03656411, -2.69975349, -2.35544645, -2.00453343, -1.64795025, -1.28667222,
   -0.92170761, -0.55409078, -0.18487501, 0.18487501, 0.55409078, 0.92170761,
    1.28667222, 1.64795025, 2.00453343, 2.35544645, 2.69975349, 3.03656411,
    3.36503877, 3.68439353, 3.99390429, 4.29291017, 4.58081629, 4.85709578,
    5.12129106, 5.37301441, 5.6119479 , 5.8378426 , 6.05051721, 6.24985614,
    6.43580706, 6.60837797, 6.76763396, 6.91369352, 7.04672474, 7.16694115,
    7.27459755, 7.36998567, 7.45342988, 7.52528284, 7.58592132, 7.63574207,
    7.67515783, 7.70459361, 7.72448306, 7.73526515, 7.73738111, 7.73127151,
    7.71737375, 7.69611969, 7.6679336 , 7.6332304 , 7.59241405, 7.54587626,
    7.49399542, 7.43713571, 7.37564643, 7.30986154, 7.24009931, 7.16666221,
    7.08983686, 7.00989417, 6.92708951, 6.84166311, 6.75384038, 6.66383247,
    6.57183674, 6.47803738, 6.38260607, 6.28570261, 6.18747563, 6.08806333,
    5.98759416, 5.88618763, 5.783955  , 5.68100008, 5.57741995, 5.47330575,
    5.3687434 , 5.26381438, 5.15859646, 5.05316445, 4.94759094, 4.84194705
])
# fmt: on
xyz = np.array([princetonD_x, np.zeros(len(princetonD_x)), princetonD_z]).T
pntslist = [Base.Vector(x) for x in xyz]

bsc = Part.BSplineCurve()
bsc.interpolate(pntslist, PeriodicFlag=False)
wire = Part.Wire(bsc.toShape())

vertexes = wire.OrderedVertexes
points = [Base.Vector(v.Point) for v in vertexes]
closure = Part.makePolygon([points[-1], points[0]])

closed_princetonD = Part.Wire([wire, closure])

shape = Part.Shape(closed_princetonD)
arc = 0
offset_princetonD = shape.makeOffset2D(0.5, arc, fill=False, intersection=True)

face = Part.Face(offset_princetonD)
hole = Part.Face(closed_princetonD)

face = face.cut([hole])

assert face.isValid()
```

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-11-08:
I believe the face produced from offset_princetonD is what is invalid here.  The cut is bad because the input to it is bad.  The offset appears to be okay, but the face made from it has defects (invalid curve on surface).

je-cook, 2024-11-08:
Thanks for looking into it @mwganson I get the same, `face.isValid() == False` obviously jumped too may steps ahead. I'll changed the title accordingly

je-cook, 2024-11-18:
I've done a bit more digging and it seems to be worked around by adjusting the tolerance on the face as suggested in this macro https://forum.freecad.org/viewtopic.php?p=409240#p409240.

If I run this on the face it becomes valid
```python
basicPrec = 0.0000001     # = Precision::Confusion()
maxGap    = basicPrec * 4.0
minEdge   = basicPrec * 2.0
rc = face.fix(basicPrec, minEdge, maxGap) 
```
It feels like this is still a regression.
I assume its because when a wire is offset any gaps increase in size so may become larger than the tolerance. This is probably dealt with in offset but when a face is created the default tolerance seems to be used. Thats my guess anyway!

luzpaz, 2024-11-26:
bumping for visibility

je-cook, 2025-08-27:
Error still present with weekly build

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/49df93a9dad1d0424c2c35b595d8ee652f35d223

It restores conversion of offset curves to B-splines before making the face, keeping the existing edge topology.