FreeCAD/FreeCAD #17709: The selection is out of date

Kique750595, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The selection is out of date, you put the cursor on a vertex or edge and select the face, you have to move the cursor back or forward until you find the exact point to select them

### Full version info

```shell
Version: 1.0.0RC2.38806 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-06:
Please try on the latest weekly and provide the full about info. Can you demonstrate this with a screenshot?

luzpaz, 2024-11-23:
@Kique750595 please try on latest 1.0.0 release. Please use full [about ](https://wiki.freecad.org/About) info. Can you perhaps make a short screencast illustrating the problem ?