FreeCAD/FreeCAD #17712: Draft_Hatch Error when trying to set 'Translate' = 'False'

semhustej, 2024-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am having a problem in attached dataset when trying to set 'Translate' = 'False' with existing Draft_Hatch object.

Steps to reproduce:

- Open file 

[FC-hatch-reduced - Copy.FCStd.zip](https://github.com/user-attachments/files/17650400/FC-hatch-reduced.-.Copy.FCStd.zip)

- Select Hatch011
- Set 'Translate' = 'False'
- Error shows up:

```
17:50:03  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39100-conda-Windows-x86_64-py311\Mod\Draft\draftobjects\hatch.py", line 135, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale * extra_scale, obj.Pattern, obj.File)
                                                     ^^^^^^^^^^^
<class 'UnboundLocalError'>: cannot access local variable 'extra_scale' where it is not associated with a value
17:50:03  Hatch011: cannot access local variable 'extra_scale' where it is not associated with a value

```

See behavior in the video:

https://github.com/user-attachments/assets/245d7d5a-59d8-4e96-bbae-9f09947b0284

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

I am not able to reproduce this issue with FC 0.21.2, so it seems like a regression to me.
I am able to reproduce with FC 1.0RC2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-06:
cc @Roy-043 (i can't test due to appimage being broken atm)

Roy-043, 2024-11-06:
Confirmed. But you need to create a new hatch.

PR #17715 (2024-11-16): Draft: Fix extra_scale error in hatch.py
Fixes #17712.
commit f723f36d (2024-11-16): Draft: Fix extra_scale error in hatch.py (#17715)

Fixes #17712.
commit c1f45b0c (2024-11-22): Draft: Fix extra_scale error in hatch.py (#17715)

Fixes #17712.
closed 2024-11-16 by Roy-043